olikraus/u8g2 #1933: ST7565 display mirror image

pierpe83, 2022-08-12:
Hi.
I am using the display with the ST7565 driver. I tried all the drivers and the one that matches is "U8X8_ST7565_ZOLEN_128X64_4W_SW_SPI u8x8" but the image is mirrored. How can I fix it?

olikraus, 2022-08-12:
One option could be to use `U8G2_MIRROR` as first argument to the constructor.
We could also create a new constructor for your display. 
Is there any datasheet for your display?

pierpe83, 2022-08-13:
Could you please explain to me where exactly and what command to use to make the display work in mirror mode? U8G2_MIRROR alone doesn't tell me much.
I do not have a datasheet for the display, but I will try to get it.

olikraus, 2022-08-13:
Oh, I just realized that you want to use U8X8. U8G2_MIRROR does not work with U8X8 (sorry for this).
If you use U8X8_ST7565_ZOLEN_128X64_4W_SW_SPI  you need to modify the code for U8X8_ST7565_ZOLEN_128X64_4W_SW_SPI  a little bit.

In your local copy of  u8x8_d_st7565.c locate lines 448 and 449

https://github.com/olikraus/u8g2/blob/08c6ff0a5666551b163aa492a41d31801f674d17/csrc/u8x8_d_st7565.c#L448-L449

Change "a1" to "a0" or "c8" to "c0". You need to figure out, which change is correct. Do not apply both changes at the same time.

pierpe83, 2022-08-13:
So the changes proposed by you do not bring any changes. The image is still displayed in a mirror image.
I noticed that other types of driver display the same image but there is a problem with the contrast, the image is barely visible. Is it possible to increase the contrast?

pierpe83, 2022-08-13:
Everything works as it should. I changed the driver to "U8X8_ST7565_EA_DOGM128_4W_SW_SPI u8x8" and set the contrast correctly. Now I know that you should handle the contrast value gently, not always 0 is the minimum setting and 255 is the maximum setting. For me, it had to be set to 30 and it's perfect.
Thank you for your help.

olikraus, 2022-08-13:
Ok, great 👍

closed 2022-08-13 by olikraus