olikraus/u8g2 #1935: which constructor use in u8g2?

Nemo79, 2022-08-13:
Which constructor can I use for this display? Selection constructor with ST7567 does not work. According to AliExpress, the given cog 12864 is the ST7565s chip.

[image]

olikraus, 2022-08-13:
Most simple solution is to just test with all of the existing u8g2 st7565 constructors. Note that it might be required to change the contrast value.

Nemo79, 2022-08-14:
sorry my mistake, chip of this display is according to AliExpress  ST7567S. display have 4 pins like on picture in previous post (sda, scl, gnd and vcc)

olikraus, 2022-08-14:
Usually there should be something to see on the display. Maybe you need to change the contrast or there is a wiring error.

Nemo79, 2022-08-15:
Hi. I think it's not in the wiring. I used 4 wires. Two for power (vcc and
gnd) and two for communication (SDA and scl). Please can you shortly tell
me if isn't error in choosing constructor with HW i2c or SW i2c? Or I must
choose SW spi or HW spi? LCD have realy only 4 pins vcc,gnd,SDA and scl.

Sorry for the many questions but I am new in Arduino world. When I ending
electrotechnical school Arduino wasn't exist. And I want to learn something
new but it's getting slowly.

Thanks a lot.

olikraus, 2022-08-15:
Use SW I2C first. Once this works, continue with HW I2C

Nemo79, 2022-08-16:
Hi Oliver, SW_I2C and HW_I2C nothing to do. I am tested it on more examples instaled with library u8g2.

So i decide attach few pictures of back side display and basic wiring.
[image]
[image]
[image]

olikraus, 2022-08-16:
How does your constructor line look like?

Nemo79, 2022-08-17:
Hi Oliver.
I am trying few constructor where was HW i2c and resolution 12864 without
reset pin.
When I come from work I make summary.

Also I tried constrtor where was SDA and scl.  But nothing changes. Display
shine but show nothing.

Tested with hello world from examples.

Nemo79, 2022-08-17:
Hi, i trying this simple code.

before compiling and uploading code to nano i uncoment (delete backslash) in constructor which i use. but nothings change

#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

//U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1306_128X64_ALT0_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);   // same as the NONAME variant, but may solve the "every 2nd line skipped" problem
//U8G2_SSD1312_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ 8);
//U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
//U8G2_ST7588_JLX12864_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ 5);

void setup(void) {
    u8g2.begin();
}   
    

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);  
}

Nemo79, 2022-08-17:
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

//U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* reset=*/ 8);
//U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display

//U8G2_SSD1312_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ 8);
//U8G2_ST7588_JLX12864_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ 5);

void setup(void) {
    u8g2.begin();
}   
    

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);  
}

olikraus, 2022-08-17:
I am confused: Initially you mentioned ST7567S controller, so why do you choose SSD1312, SSD1306, ST7588 constructors? They will not work. You need to use a ST7567 constructor if your display includes the ST7567 controller. 
Try one of these:
```
U8G2_ST7567_JLX12864_F_SW_I2C
U8G2_ST7567_ENH_DG128064_F_SW_I2C
U8G2_ST7567_OS12864_F_SW_I2C
```

Also try to avoid SDL and SCL macros. You never know how they are defined. You have connected your board to A5 and A4, so use these pins with their names. For example it could look like this:
```
U8G2_ST7567_JLX12864_F_SW_I2C u8g2(U8G2_R0, /* clock= */ A5, /* data= */ A4, /* reset=*/ U8X8_PIN_NONE); 
```

Nemo79, 2022-08-18:
I'm confused by that too. I tried the possible and the impossible, but it doesn't work.
in the description https://www.aliexpress.com/item/1005004252465006.html?spm=a2g0o.order_list.0.0.21ef1802fkwjG3

ST7567S or Equal is listed

olikraus, 2022-08-18:
> ST7567S or Equal is listed

Super, so, did the constructor 

``` 
U8G2_ST7567_JLX12864_F_SW_I2C u8g2(U8G2_R0, /* clock= */ A5, /* data= */ A4, /* reset=*/ U8X8_PIN_NONE); 
```

work?

Nemo79, 2022-08-18:
'U8G2_ST7567_JLX12864_F_SW_I2C' does not name a type; did you mean 'U8G2_ST7588_JLX12864_F_SW_I2C'?

olikraus, 2022-08-18:
There should be the controller name in the constructor name: ST7567

Nemo79, 2022-08-18:
now on my ntb i am trying this constructor which you send and no error apears. but maybe i must play a litle while with contrast because i dont see anything on display. compilation and uploading was ok

olikraus, 2022-08-18:
For me also the `U8G2_ST7567_JLX12864_F_SW_I2C` works without problems.
But looks like you have solved the compile error..

Anyhow, beta is here:

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

Nemo79, 2022-08-20:
Hi Oliver, in my first testing code hello world was problem with i2c
address of display. After setting corect i2c address display show
characters from hello world example.

olikraus, 2022-08-20:
Ok, so your problem is solved?

Nemo79, 2022-08-21:
yes it is solved. thank you very much.

olikraus, 2022-08-21:
:)

closed 2022-08-21 by olikraus