olikraus/u8g2 #1937: 80ms to render one page?

gary00k, 2022-08-21:
Hello,
I use ST7567 128x64px (constructor U8G2_ST7567_JLX12864_F_4W_HW_SPI).
Is it normal that rendering one page takes ~80ms on Atmega644 @ 18,432MHz?

Is there any way to speed it up?

olikraus, 2022-08-21:
Depends an your code...

gary00k, 2022-08-21:
For example:

do {
     u8g2.setFont(u8g2_font_p01type_tn);

      u8g2.drawButtonUTF8(5, 39, U8G2_BTN_BW1, 0,  1,  1, "1/32");
      u8g2.drawButtonUTF8(26, 39, U8G2_BTN_BW1, 0,  1,  1, "1/16");
      u8g2.drawButtonUTF8(45, 39, U8G2_BTN_BW1, 0,  1,  1, "1/8");
      u8g2.drawButtonUTF8(62, 39, U8G2_BTN_BW1, 0,  1,  1, "1/4");
      u8g2.drawButtonUTF8(79, 39, U8G2_BTN_BW1, 0,  1,  1, "1/2");
      u8g2.drawButtonUTF8(96, 39, U8G2_BTN_BW1, 0,  1,  1, "3/4");
      u8g2.drawButtonUTF8(115, 39, U8G2_BTN_BW1, 0,  1,  1, "1/1");

      u8g2.setFont(u8g2_font_pcsenior_8r);
      u8g2.drawStr(5, 52, "TIME  666ms");
      u8g2.drawStr(5, 62, "PARAM 69%");
      u8g2.drawStr(5, 62, buffer);
  } while ( u8g2.nextPage() );

Render time: ~82ms.

gary00k, 2022-08-21:
For example:

```
do {
     u8g2.setFont(u8g2_font_p01type_tn);

      u8g2.drawButtonUTF8(5, 39, U8G2_BTN_BW1, 0,  1,  1, "1/32");
      u8g2.drawButtonUTF8(26, 39, U8G2_BTN_BW1, 0,  1,  1, "1/16");
      u8g2.drawButtonUTF8(45, 39, U8G2_BTN_BW1, 0,  1,  1, "1/8");
      u8g2.drawButtonUTF8(62, 39, U8G2_BTN_BW1, 0,  1,  1, "1/4");
      u8g2.drawButtonUTF8(79, 39, U8G2_BTN_BW1, 0,  1,  1, "1/2");
      u8g2.drawButtonUTF8(96, 39, U8G2_BTN_BW1, 0,  1,  1, "3/4");
      u8g2.drawButtonUTF8(115, 39, U8G2_BTN_BW1, 0,  1,  1, "1/1");

      u8g2.setFont(u8g2_font_pcsenior_8r);
      u8g2.drawStr(5, 52, "TIME  666ms");
      u8g2.drawStr(5, 62, "PARAM 69%");
  } while ( u8g2.nextPage() );
```

Render time: ~82ms.

olikraus, 2022-08-21:
Looks ok. You could change some settings in u8g2.h but I think there is nothing much to improve.

gary00k, 2022-08-21:
Well, I deleted my post bacause I attached not a full code, my bad.
Btw, is there any possibility to "rerender" only a part of frame? For example, only one character.

strud, 2022-08-21:
@gary00k I have been writing/updating only parts of the screen required to change using the buffer method

eg, wiping out previous text with a rectangle then writing the nex text, then sending the buffer.

`u8g2.setDrawColor(0);// black
          u8g2.drawBox(85, 14, 40, 10);
          u8g2.setDrawColor(1);// change to white to write text
          u8g2.setCursor(85, 24);
          u8g2.setFont(u8g2_font_timB08_tr);
          u8g2.printf("%3d", ival);
          u8g2.sendBuffer();`

olikraus, 2022-08-21:
There is also the UpdateArea function, see the manual and examples for partial update.

closed 2023-02-22 by olikraus