olikraus/u8g2 #1938: Add Allvision QG-9632TLBZG02 support

grogcw, 2022-08-23:
Hi !

I salvaged a screen from a device and I'm trying to upcycle it.

It's an Allvision QG-9632TLBZG02, 14pin, 96x32 pixels using SSD1316 driver (wiring is the same as 128x32 screens).

The closest match to this in the library is U8G2_SSD1316_128X32, and is currently working for my tests but sometimes I get problems displaying fonts resulting in a blank screen.

I had success drawing shapes using a bitmap converted to bytes corresponding to its size (which is flipped between MSB and LSB tho).

Here's a link to it's [datasheet](https://datasheet.lcsc.com/lcsc/1804162033_Shenzhen-Allvision-Tech-QG-9632TLBZG02_C91216.pdf).

Can you please add it to the library ?

Cheers !

olikraus, 2022-08-23:
I don't have access to the repo at the moment, but I can add the constructor later. Can you give more details on the black screen problem?

grogcw, 2022-08-23:
The problem seems to occur when I use different fonts, some work, some don't.

I could dress up a list if you want, but I'm quite convinced that it's size related.

olikraus, 2022-08-23:
A simplified not working example would be great.

grogcw, 2022-08-23:
The following exemple gives me a blank screen.

```
#include <Arduino.h>
#include <Wire.h>
#include <U8g2lib.h>

U8G2_SSD1316_128X32_2_HW_I2C u8g2(U8G2_R0, 21, 20);

void setup() {
  u8g2.begin(); //Start Screen
}

void loop() {
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_10x20_tn);
  u8g2.drawStr(0, 20, "Hello World !");
  u8g2.sendBuffer();
  delay(1000);
}
```
In a second time, if I reduce the text's length, with "01:04" for exemple, [the text get cropped in the middle](https://imgur.com/a/anwtyTJ).

However, if I change the offset to 14 in y, it's shown properly but that doesn't resolve the problem with longer strings.

Any idea ?

olikraus, 2022-08-24:
Replace the "2" in the constructor with "F".

olikraus, 2022-08-24:
In other words, please check between full and page mode in u8g2.

grogcw, 2022-08-30:
I'll check it in a few weeks for sure, I have an heavy schedule ahead !

Thanks for the help anyway :)

olikraus, 2022-08-30:
I have created beta release 2.33.12. It includes the new constructor "U8G2_SSD1316_96X32_2_HW_I2C" (and others)

Please execute u8g2 page buffer flip mode example. There should be a frame at the border of the display which should be there all the time if everything is correct.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

grogcw, 2022-09-27:
I can now safely confirm that everything works as planned !

Thanks for your support,

Cheers !

olikraus, 2022-09-29:
Great, thanks for the feedback.

closed 2023-02-22 by olikraus