olikraus/u8g2 #1939: i2c support was removed for HEM6432

obaterspok, 2022-08-28:
Hi,

In commit 
https://github.com/olikraus/u8g2/commit/5b611bc602eccb07b3b1e116808900903a86d17a

The i2c support was removed for HEM6432. Any reason for this?

-- john

olikraus, 2022-08-28:
Hmm I don't remember. I do not have proper access to the repo, so I will check this in the next days.

olikraus, 2022-09-03:
not sure why this has happend... will be fixed...

olikraus, 2022-09-03:
I have created beta release 2.33.13, maybe you can confirm that this works again...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

olikraus, 2022-09-04:
I will close this for now, nevertheless any feedback on this is welcome.

closed 2022-09-04 by olikraus

obaterspok, 2022-09-04:
Works perfect with esp32-c3 for both sw & hw i2c.

U8G2_ST7567_HEM6432_1_SW_I2C u8g2(U8G2_R2, PIN_SCL, PIN_SDA, U8X8_PIN_NONE);
U8G2_ST7567_HEM6432_1_HW_I2C u8g2(U8G2_R2, U8X8_PIN_NONE, PIN_SCL, PIN_SDA);

Thanks a lot!

olikraus, 2022-09-04:
Great, thanks a lot for pointing out this issue and for your feedback.