olikraus/u8g2 #1941: RLE Compression , repeat number

IronMan-Nam, 2022-08-30:
First, I am very grateful you have created a great library to ease the development of our projects.
after learning and experimenting. I saw one thing that I think we should change it.
In the source file bdf_rle.c function bg_prepare_01_rle(bg_t *bg, unsigned a, unsigned b). I change the comparison operand from >= in the two while loops to >.
That reduces one repetition if the length of bit 0 or bit 1 is within the specified number of bits.
Example in case:
bg->rle_bits_per_0= 3,
bg->rle_bits_per_1 = 4,
a = 7, b = 4
The number of compressions is 1 time instead of 2 times before the improvement.

olikraus, 2022-08-30:
You refer to this code, right?
https://github.com/olikraus/u8g2/blob/7e58b87b3cd63ed4016624cd00226ffe4f0bfe1d/tools/font/bdfconv/bdf_rle.c#L339-L359

Im instead if changing >= to > we could also remove the "-1" right?

IronMan-Nam, 2022-09-01:
Yes. That right.

IronMan-Nam, 2022-09-01:
don't need remove -1. 
Because I think it possible with my case, the case that I mentioned above.

compressed bits instead of b0111 0000 0000 0100
After changing >= to >. it will become b0111 0100

olikraus, 2022-09-02:
Some results:
new: u8g2_font_9x15_tf[2606]
old: u8g2_font_9x15_tf[2657]
--> 2% reduction
new: u8g2_font_tom_thumb_4x6_te[1483]
old: u8g2_font_tom_thumb_4x6_te[1515]
--> 2.2% reduction

new: u8g2_font_t0_11b_te[7918]
old: u8g2_font_t0_11b_te[8233]
--> 3.8% reduction

The above example are kind of "good". I see many fonts which have lesser or none reduction.
So I would say, the reduction is up to 4% but will be usually around 1%. Still good... But I now need to check whether the results are still the same.

olikraus, 2022-09-02:
After reviewing the font decoding results: All seems to be ok. 

@IronMan-Nam I am really impressed you figured out that bug? How did you found it?

olikraus, 2022-09-03:
todo for next u8g2 release; copy png files from build folder to wiki img dir.

olikraus, 2022-09-03:
I have created beta release 2.33.13....

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

IronMan-Nam, 2022-09-03:
> After reviewing the font decoding results: All seems to be ok.
> 
> @IronMan-Nam I am really impressed you figured out that bug? How did you found it?

I want to study your source code in data compress. I have debugged each function to understand their principles. Although I have little experience and take a long time to do it, I feel very excited. fortunately I found this bug.

Hope you can develop more code sources.

olikraus, 2022-09-03:
wow, I never thought someone could be interested in the compression algorithm.

But yes, good finding. Thanks again.

closed 2022-09-03 by olikraus