olikraus/u8g2 #1943: RAM overflow with Microchip Studio and SAMD21 device

MilouBlue, 2022-09-02:
Hello there,

I am trying to implement the u8g2 library with i2c under Microchip Studio in a SAMD21 device. I mainly used [this youtube video](https://www.youtube.com/watch?v=TO7-yxT6Gvw) for avr devices as a reference, which was linked in the wiki. The code compiles but it results in a ram overflow of 142536 bytes. I don’t seam to find the reason for the huge ram usage and hope someone has an idea. The error messages are:

```
samd21g18a_test.elf section `.bss' will not fit in region `ram'
region `ram' overflowed by 142536 bytes

```
I configuered the 8MHz internal oscilator as clock with a prescaler of 2, so 4MHz. I added the symbol `F_CPU=4000000` as discribed in the tutorial.
I have reduced my code to only the setup of the display. So it does not have to do with fonts or anything, as I am not using that yet. If the setup is commented out the size of ram usage returns to normal. I tried to trace back the implementations of the setup function, but could not find anything. I am not too experienced with that kind of programming, so not sure.  Changing the display in use does not change the situation.
This is my main:

```
#include <atmel_start.h>

#include <u8g2.h>
#include "display/display_implementations.h"

u8g2_t u8g2;

int main(void)
{
	atmel_start_init();

  u8g2_Setup_ssd1306_i2c_128x32_univision_1(
  &u8g2, U8G2_R0,
  u8x8_byte_i2c,
  u8x8_delay
  );
  
/*
  u8g2_InitDisplay(&u8g2);
  u8g2_SetPowerSave(&u8g2, 0);
	  u8g2_ClearBuffer(&u8g2);
	  u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
	  //u8g2_DrawStr(&u8g2, 0, 0, "");
	  u8g2_SendBuffer(&u8g2);
*/
	while (1) {
	}
}
```

I made my own version of the i2c-interface and the delay function. I am using the drivers provided by atmel start.
The `u8x8_byte_i2c` is basically a copy of what I found in the wiki with the i2c write function added and  `u8x8_delay` is adapted from the avr tutorial.

```
uint8_t u8x8_byte_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	static uint8_t buffer[32];		/* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
	static uint8_t buf_idx;	
	uint8_t *data;
	
	switch(msg)
	{
		case U8X8_MSG_BYTE_SEND:
		data = (uint8_t *)arg_ptr;
		while( arg_int > 0 )
		{
			buffer[buf_idx++] = *data;
			data++;
			arg_int--;
		}
		break;
		case U8X8_MSG_BYTE_INIT:
		/* add your custom code to init i2c subsystem */
		break;
		case U8X8_MSG_BYTE_SET_DC:
		/* ignored for i2c */
		break;
		case U8X8_MSG_BYTE_START_TRANSFER:
		buf_idx = 0;
		break;
		case U8X8_MSG_BYTE_END_TRANSFER:
		i2c_m_sync_set_slaveaddr(&I2C_0, SSD1306_091_I2C_ADDR, I2C_M_SEVEN);
		i2c_m_sync_write_(&I2C_0, buffer, buf_idx); // duplicate of the atmel start function which is not declared outside the .c
		break;
		default:
		return 0;
	}
	return 1;
}

uint8_t u8x8_delay (u8x8_t * u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
	// uint8_t cycles;

	switch(msg) {
		case U8X8_MSG_DELAY_NANO:  // delay arg_int * 1 nano second
		// At 20Mhz, each cycle is 50ns, the call itself is slower.
		break;
		case U8X8_MSG_DELAY_100NANO:       // delay arg_int * 100 nano seconds
		break;
		case U8X8_MSG_DELAY_10MICRO: // delay arg_int * 10 micro seconds
		while( arg_int-- ) delay_us(10);
		break;
		case U8X8_MSG_DELAY_MILLI:  // delay arg_int * 1 milli second
		while( arg_int-- ) delay_ms(1);
		break;
		default:
		return 0;
	}

	return 1;
}
``` 

I am happy for any help and ideas!

olikraus, 2022-09-03:
Maybe the compile flags `-ffunction-sections` `-fdata-sections` and the linker flag `-Wl,--gc-sections` are all or partly missing.

MilouBlue, 2022-09-03:
That was it. The `-fdata-sections` was missing. It compiles now without errors and I can go on to the actual implementation.
Thank you very much for the fast help, I wouldnt have found that alone.

olikraus, 2022-09-03:
🤗

closed 2023-02-22 by olikraus