olikraus/u8g2 #1944: SH1106 issue with userInterfaceSelectionList

HARA2020, 2022-09-03:
I am trying to use the **userInterfaceSelectionList()** feature but am having problems with the next, previous, up, and down inputs as called out in the **begin()** function.  These inputs are tied to a switches that when activated, grounds the input.  There is a pullup on the line.  Activating the switch moves the menu item but releasing the switch moves it as well.  It seem to be triggering on both the falling and rising edge.
I have to use u8g2 v2.32.15 as the most current version does not work at all.  I am using the following constructor:
U8G2_SH1106_128X64_NONAME_F_HW_I2C.  
I am using Visual Code/Platform I/O, Arduino development environment with a STM32 Blue Pill.
My software expertise is somewhat lacking so I apologize in advance if there is something I am missing in the documentation.

olikraus, 2022-09-03:
Any code example?

HARA2020, 2022-09-03:
From the platformio .ini file
lib_deps =
    olikraus/U8g2 @ 2.32.15

// Used for the switches
const int ENCSW = PB1;          //Encoder switch
const int CLEARSW = PB15;
const int ARMSW = PB14;

From main()

// OLED Display
U8G2_SH1106_128X64_NONAME_F_HW_I2C mainDisplay(U8G2_R0, U8X8_PIN_NONE)

pinMode(ENCSW, INPUT_PULLUP);                     
pinMode(CLEARSW, INPUT_PULLUP);
pinMode(ARMSW, INPUT_PULLUP);

// OLED Display setup
u8g2_SetI2CAddress(&mainDisplay, 0x3C); // Setup the 1.3" OLED display
mainDisplay.setBusClock(800000);
mainDisplay.begin(ENCSW,U8X8_PIN_NONE,U8X8_PIN_NONE,ARMSW,CLEARSW,U8X8_PIN_NONE);

unsigned int configMenu(void)
{
  mainDisplay.clear();
  mainDisplay.setFont(u8g2_font_ncenB08_tr);
  return(mainDisplay.userInterfaceSelectionList("Configuration Menu", 1, "Set Cont. Address\nSet Bank Address\nSet Bank RF Channel\nSet Cont. RF Channel\nDone"));
}
[main.zip](https://github.com/olikraus/u8g2/files/9483214/main.zip)

olikraus, 2022-09-03:
Hmmm looks good. Maybe an hardware issue?

```
u8g2_SetI2CAddress(&mainDisplay, 0x3C); // Setup the 1.3" OLED display
```
should better be
```
mainDisplay.setI2CAddress(0x3C); // Setup the 1.3" OLED display
```

olikraus, 2022-09-03:
There is a bug in the current release, which prevents the event detection to work: https://github.com/olikraus/u8g2/pull/1929

I have reverted the pull request.

olikraus, 2022-09-03:
Actually this bug was introduced in 2.33.something, but version 2.32.15 should be ok.

olikraus, 2022-09-03:
I have created beta release 2.33.14, which should work.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

HARA2020, 2022-09-04:
Thank you.  Adding the lib_deps worked somewhat.  My original problem remains.  Activating the switch moves the menu item but releasing the switch moves it as well. It seem to be triggering on both the falling and rising edge.  The switch is a momentary SPDT (center off) switch.  The center is connected to ground.  Activating the switch grounds the digital input it is connected to.

olikraus, 2022-09-04:
Strange, for me it works as expected with Arduino Uno.

HARA2020, 2022-09-05:
I did a test program (attached) and bread boarded the circuitry using a new Blue Pill (STM32F103C8) and new display.  The result is the same.  I do not know how you are reading ports PB14 and PB15.  It seems that the code triggers on a falling and rising edge.  (Activating a switch brings the input low).  Are you using interrupts or something that triggers on edges?

[main.zip](https://github.com/olikraus/u8g2/files/9485774/main.zip)

HARA2020, 2022-09-05:
OK so I disconnected the serial port monitor and removed the SWD programming cable and now it works sometimes.  I discovered that adding a decoupling capacitor (.01uf) to the switch inputs stabilizes the operation.  I am going add them to my project and try it again.

HARA2020, 2022-09-05:
Adding capacitors to the switch inputs did the trick.  Thanks for your help and for identifying the bug with the library.

closed 2023-02-22 by olikraus