olikraus/u8g2 #1949: SPI 3 wires in stm32

leonardowolf, 2022-09-06:
Hello there,

I'm trying my best to use the library, but i can't figure out how to properly send data to the SSD1322 display i'm using since the communication is SPI 3 wires but in hardware. I did the right configuration, so the protocol send 9 bits and i believe i did the right clock, phase, parity configuration on the spi setup. But my display still dont show nothing.
Here is my code
https://github.com/leonardowolf/f302_nhdinitializer
the constructor can be found here:
https://github.com/leonardowolf/f302_nhdinitializer/blob/main/Core/Src/ac.c

any help would be appreciated

olikraus, 2022-09-06:
At least the other delay messages need to be implemented:
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#template-for-the-gpio-and-delay-callback

3-wire SPI probably is the most untested protocol with u8g2. If possible start with 4-wire SPI first.

leonardowolf, 2022-09-08:
i have already used this lib with 4 wires and it worked like a charm.

leonardowolf, 2022-09-08:
i updated my code with some delay messages that where missing. Didn't work. My best guess is that since I'm using 3 wires config, the display is waiting for a 9bit packet where the  9nth is the D/C state. But the MSG system calls the case on the "delay and gpio function"
'
	case U8X8_MSG_GPIO_DC:
		break;
'
this one to be precise. This call is understandeable on the 4wires config bc i need to pass the uc pin to recive the arg_int value. But in the 3 wire case there shoud be a way to embed this variable in the packet witch wil be sent to the SSD1322 via spi

olikraus, 2022-09-08:
Ah, yes, you are right, The C interface does not include the 9bit transmission.
Like you mentioned, you need to listen to the MSG_GPIO_DC and remember the arg_int value as DC bit. Then you have to construct an array with 9 bit code words which are then transmitted. I see two options:
 * You controller supports 9 bit SPI transmission. In this case you need to store the value of the last MSG_GPIO_DC, prepend the actual 8 bit data payload with this additional bit and send this byte.
 * If your controller only supports 8 bit SPI, then you need to convert the 9 bit stream of code words into an 8 bit array.
 
The second approach is used for Arduino, but is only available in C++ interface:
https://github.com/olikraus/u8g2/blob/a6843a12e1a7388f32eac88b832c25310e99a508/cppsrc/U8x8lib.cpp#L710-L866

I think you could mostly copy that code if you want to follow the approach. You just need to replace the Arduino SPI function calls with your own code.

dolence, 2022-09-16:
I've checked the initialization of SPI and all the registers seem to be right but unfortunately the display doesn't initialize. Could it be related to the fact my display doesn't useall the four wires? It doesn't use the DC pin. I could give the Arduino version a try and see if this display work in 3-wire mode...

ntchris, 2022-09-29:
feel free to take a look at my example here

I spent 5 days to make it work.

bluepill board, STM32F103C8T6,  hardware SPI, 
using STM32CUBEIDE ,  Using C++ style of U8G2.
LCD12864 ( ST7920 )   ( a regular one, even it works with 3.3V stm32 )
It has no DC, I think for most of the screen type, CS can always be pulled up(or down , depends) to make it easier, therefore no start_transfer or end_transfer function needed)
(when using 5V peripheral,  one must be very careful about unpowered stm32 being damaged by a 5V peripheral or signal. )

https://github.com/ntchris/stm8_stm32/blob/master/bluepill_LCD12864_u8g2/Core/Src/stm32_u8g2_lcd12864.cpp

`
this is the key step.
uint8_t u8x8_byte_3wire_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
    switch (msg) {
    case U8X8_MSG_BYTE_SEND:
       HAL_SPI_Transmit(&LCD_SPI, (uint8_t*) arg_ptr, arg_int, 10000);
        break;`

olikraus, 2022-09-29:
> I think for most of the screen type, CS can always be pulled up(or down , depends) to make it easier, therefore no start_transfer or end_transfer function needed)

Maybe, but CS is also used to reset the communication so that the display knows bit zero during transfer. The CS acts like the start bit in UART communication.

closed 2023-02-22 by olikraus