olikraus/u8g2 #1953: u8g2 and m2tklib compatibility

dolence, 2022-09-12:
Hi! I have used u8g and m2tklib on a barebone avr chip. Is u8g2 and m2tklib compatible?

olikraus, 2022-09-12:
No, I never ported m2tklib to u8g2.
Instead there is MUI, which will be part of u8g2.

dolence, 2022-09-12:
> 

Is MUI usable in any form? I would like to test it if possible.

olikraus, 2022-09-12:
> Is MUI usable in any form? 

I would say yes

https://github.com/olikraus/u8g2/wiki/muiref

https://github.com/olikraus/u8g2/wiki/muimanual

https://github.com/olikraus/u8g2/issues/1747

Examples are here (See MUI prefix):
https://github.com/olikraus/u8g2/tree/master/sys/arduino/u8g2_page_buffer

MUI vs m2tklib

MUI is simpler
MUI requires lesser memory
MUI does not include input devices, instead 3rd party libs have to be mapped to MUI events (see examples)
MUI is part of and will only support u8g2

dolence, 2022-09-12:
MUI seem to be just what I need. Thank you very much! 
Actually I'm following your Atmel Studio guide to test this ST7920 in bit banging SPI mode. Any tips to use HW SPI instead?

closed 2022-09-12 by dolence