olikraus/u8g2 #1955: ESP32-PICO-D4 and SH1106

h-RAT, 2022-09-16:
Hi, I use an electronic card controlled by an ESP32-PICO-D4 with a SH1106 128X64 I2C display.

The display is connected on pins 33 and 15.

So I tried this:

- U8G2_SH1106_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, 33 , 15, U8X8_PIN_NONE);

or

- U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, 33, 15, U8X8_PIN_NONE);

I also tried adding this:

- Wire.begin(33, 15);

And it doesn't work.

How can I use your library?

(Works with Adafruit_SSD1306 and Wire.begin(33 , 15))

Thanks

olikraus, 2022-09-16:
> U8G2_SH1106_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, 33 , 15, U8X8_PIN_NONE);

should work and "Wire.begin(33, 15);" would not be required for this.

Did you use any existing u8g2 example?

h-RAT, 2022-09-17:
> 

Yes, I use the "Hello World" example.

I'm familiar with Arduino and libraries, but this time I don't understand where is the problem.

My display works fine with another library so it's not the display or the ESP the problem.

Thanks

olikraus, 2022-09-17:
Hmmm I think for hw i2c the reset pin is second argument

h-RAT, 2022-09-18:
> Hmmm I think for hw i2c the reset pin is second argument

I tried this:

- U8G2_SH1106_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, 33 , 15, U8X8_PIN_NONE);
- U8G2_SH1106_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, 15, 33, U8X8_PIN_NONE);
- U8G2_SH1106_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, 15, 33);
- U8G2_SH1106_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, 33, 15);

- U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, 33 , 15, U8X8_PIN_NONE);
- U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, 15, 33, U8X8_PIN_NONE);
- U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, 15, 33);
- U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, 33, 15);

Nothing works :/

olikraus, 2022-09-18:
My suggestion is to continue with `U8G2_SH1106_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, 33 , 15)` if 33 is the clock input of your display, otherwise use `U8G2_SH1106_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, 15, 33)`.

If Adafruit Lib works exactly with the same hardware setup then also the above constructor should work.

Do you have any other devices on the same I2C bus?

Maybe you can also sent a picture of your setup.

h-RAT, 2022-09-18:
- U8G2_SH1106_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, 33 , 15)
- U8G2_SH1106_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, 15, 33)

Still does not work :/

Yes the Adafruit library works perfectly with the same configuration, in any case I can't change anything, it's a card with an ESP32-PICO-D4 type SMD. All welds are made by the manufacturer.

No other device connected in I2C, only 3 devices connected in SPI.

Yes I can potentially send you a photo in private (It's a "prototype card" not yet marketed, I don't think I can release a photo in public).

Thanks

olikraus, 2022-09-18:
Well, I don't know. I do not see a reason why Adafruit lib works, but u8g2 fails. 
Are you able to see any activity on the two pins?
Is the ESP32 Arduino base library ported correctly to the PICO-D4?

Actually I don't know what to suggest here.

h-RAT, 2022-09-18:
I will try to see if there is activity on the PINs, for the moment I don't have the necessary equipment.

Well except the problem with u8g2 I have no problem with the ESP32-PICO-D4, everything seems to work perfectly.

I myself don't understand why it not work, in theory everything should work.

Thanks for taking the time to help me.

closed 2023-02-22 by olikraus