olikraus/u8g2 #1956: U8X8_HAVE_2ND_HW_SPI, using the 2nd SPI port of a Teensy 4.1

WerkstattKreuzberg, 2022-09-17:
Hello Oli,

i have connected  a SSD1306 to the 2nd SPI port of a Teensy 4.1 (SPI1):
U8G2_SSD1306_128X64_NONAME_F_2ND_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 33, /* dc=*/ 22, /* reset=*/ 23);
The display has no cs pin, so the 33 is just a dummy. But it works.
A first test was not successful. Digging around showed that U8X8_HAVE_2ND_HW_SPI will not defined.

In the U8x8lib.h i found:
/* define U8X8_HAVE_2ND_HW_SPI if the board has a second wire interface*/
/* As of writing this, I did not found any official board which supports this */
/* so this is not tested (May 2017), issue #224 */
/* fixed ifdef, #410, #377 */
/* meanwhile it is defined e.g. here: https://github.com/arduino/ArduinoCore-samd/blob/master/variants/mkrzero/variant.h#L91 */
/* so it should be available for mkrzero */

#ifdef SPI_INTERFACES_COUNT
#if SPI_INTERFACES_COUNT > 1
#define U8X8_HAVE_2ND_HW_SPI
#endif
#endif

As it seems SPI_INTERFACES_COUNT is not used by Teensy's SPI library. So the workaround for me is to edit
U8x8lib.h and just add
#define U8X8_HAVE_2ND_HW_SPI

With the explicit define the display works great at the 2nd SPI port.
But is there perhaps a more elegant variant that you could put into the library? :-)

Thank you for all the work!

olikraus, 2022-09-17:
SPI_INTERFACES_COUNT is used by the libs for the official Arduino Boards

WerkstattKreuzberg, 2022-09-17:
Ok so we have to use this little  workaround.

closed 2022-09-17 by WerkstattKreuzberg