olikraus/u8g2 #1957: support HD66753 driver

jaksatomovic, 2022-09-17:


jaksatomovic, 2022-09-17:
http://janvitek.org/vitekj/490s11/Resources_files/HD66753e.pdf
for this lcd 
https://www.beyondinfinite.com/lcd/Library/Optrex/F-51824FNF-SLW-AE.pdf

olikraus, 2022-09-17:
Looks different to anything what I ever had implemented. Looks like a new serial communication procedure would be required.
Is there a display with this controller available for sale somewhere?

jaksatomovic, 2022-09-17:
It is actually monochrome display from ipod classic 4th gen

jaksatomovic, 2022-09-17:
Would something like this 

https://github.com/hossboss/SPI_Display_Test/tree/master/source/HD66753

with some extra modifications work?

olikraus, 2022-09-17:
so you know the pinout of the ipod display?

> with some extra modifications work?

helpful, but still a lot of work

jaksatomovic, 2022-09-18:
Correct, i know the pinout. 
Hmm ok then, thx anyway.
It looks like i wont be using ipod’s display in near future 😅

olikraus, 2022-09-18:
I probably would need the pinout of the display connector and also the display itself. Is there any place to buy a broken ipod (with  a working displayI mean, I don't have time at the moment, but maybe if I get access to pinout and display...

jaksatomovic, 2022-09-18:
here can be found pinout for the hd66753: https://www.egr.msu.edu/classes/ece480/capstone/spring15/group03/docs/ECE480ApplicationNote.pdf
There is a lot broken ipods on ebay, also they can be found on aliexpress. 
Well i am not sure where are you from but i would happily donate my lcd although i am not sure what would be shipping costs from Croatia :)

olikraus, 2022-09-18:
Thanks for the pinout. At least at the moment, I don't have time to port u8g2 to that display. It also seems to be difficult to find such a display in internet for sale. So for users who need a display, there seem to be cheaper options. Maybe i once see an offer for a broken ipod classic ;-)
Greetings from Germany

jaksatomovic, 2022-09-18:
Ok then, thanks for your time and effort anyway. Let me know if anything comes up.
 Best regards from Croatia

closed 2022-09-18 by jaksatomovic