olikraus/u8g2 #1959: SSD1306 96X39

fastfourier666, 2022-09-17:
Hi!

I have some of [these displays](https://www.buydisplay.com/96x39-pixel-0-83-inch-small-oled-display-manufacturers-i2c-serial-spi).

I did try the `U8G2_SSD1306_96X40_1_HW_I2C` constructor but it skips every other line. The `U8G2_SSD1306_128X64_NONAME_1_HW_I2C` displays correctly, but the 0,0 point is way off the corner of the screen.

Is easy to add support for this one? Or any pointers on how to create one myself?

Thanks!

olikraus, 2022-09-18:
How will your displays work with `U8G2_SSD1306_96X40_1_HW_I2C` constructor?

> but the 0,0 point is way off the corner of the screen.

You could change the offset for your display in your local copy of the init sequence: https://github.com/olikraus/u8g2/blob/a6843a12e1a7388f32eac88b832c25310e99a508/csrc/u8x8_d_ssd1306_128x64_noname.c#L50

> Is easy to add support for this one? 

If the above 96x40 works, then I could just create a copy of that constructor...

fastfourier666, 2022-09-19:
Yes I did try the `U8G2_SSD1306_96X40_1_HW_I2C` since it's closest to my display size, but I get the every other line problem: 

```
#include <Arduino.h>
#include <U8g2lib.h>

#include <Wire.h>

U8G2_SSD1306_96X40_1_HW_I2C u8g2(U8G2_R0, 5);

void setup() {
  // put your setup code here, to run once:
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();          // clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr); // choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");  // write something to the internal memory
  u8g2.sendBuffer();          // transfer internal memory to the display
  delay(1000);  
}
```

[image]

Thanks for the tip about the init sequence. I think I'll just do that for now. I was afraid I'd just be wasting memory on a bigger screen buffer, since I will be running 12 of these displays off one ESP32.

olikraus, 2022-09-19:
Instead of changing the original file, you could also send commands directly to the display with the "u8g2.sendF()" (https://github.com/olikraus/u8g2/wiki/u8g2reference#sendf) command.
The alternating line topic is configured with the "0xda" command (see the ssd1306 datasheet).
It probably should look like this after the .begin() command:
```
u8g2.sendF("ca", 0xda, 0x02);
```
Maybe you need to add a different y offset:
```
u8g2.sendF("ca", 0xd3, offset);
```

fastfourier666, 2022-09-19:
Hi. Thanks for your detailed reply. When I was searching for this "every other line" issue, there are several posts out there that claim that it's a result of a broken display controller. At least now I know my gear is working OK!

Using the `U8G2_SSD1306_96X40_1_HW_I2C` constructor I have tried playing around with various combinations of commands `0xC0`/`0xC8`, and setting bits 4 and 5 of `0xDA` but I can't get 0,0 to appear at the top. 

I can change the Y offset so Y=0 appears at the top row of the screen, but the left column of pixels are always X=32.

I could probably deal with this but half the screens in my project need to be rotated 180, and when I do that the drawing positions are different...

fastfourier666, 2022-09-19:
I read the FAQ and thought this could be a SH1106, even though the manufacturer says SSD1306. `U8G2_SH1106_128X64_NONAME_1_HW_I2C` works a little better - now the columns X=0 and X=1 are off the side of the screen. X=2 is the leftmost column of visible pixels.

Is there any way of figuring out which controller I have?

At any rate, I would still like to get the rotation working properly so I don't have to draw the screen two different ways.

Thanks for your continued support!

olikraus, 2022-09-20:
> Is there any way of figuring out which controller I have?

Difficult in general. The SH1106 has 132 columns, the SSD1306 only 128.

> but the left column of pixels are always X=32.

It can be configured: 
https://github.com/olikraus/u8g2/blob/a6843a12e1a7388f32eac88b832c25310e99a508/csrc/u8x8_d_ssd1306_128x64_noname.c#L353

>  I would still like to get the rotation working properly

Let me know which additional commands are required for your display (sendF), then I will create a constructor for your display with the correct dimensions. After that we will work on the rotation.

fastfourier666, 2022-09-20:
Thanks very much!

I changed the default_x_offset to 32
and do `u8g2.sendF("ca", 0xd3, 39);` in the setup.

Now I can do `u8g2.drawFrame (0,0,96,39);` and it fills the display. Great!

fastfourier666, 2022-09-20:
Thanks very much!

On line 353, I changed the default_x_offset to 32
and do `u8g2.sendF("ca", 0xd3, 39);` in the setup().

Now I can do `u8g2.drawFrame (0,0,96,39);` and it fills the display. Great!

fastfourier666, 2022-09-21:
I did a couple more experiments regarding the rotation.

Using `U8G2_R0` the coordinates are correct and `u8g2.drawFrame (0,0,96,39);` draws a box around the edge of the display.

With `U8G2_R2`, I get the same box with `u8g2.drawFrame (32,25,96,39);`

Thank you

olikraus, 2022-09-21:
If the rotation is setup correctly, then the drawFrame command should have the same arguments for R0 and R2.

So I will use U8G2_SSD1306_96X40 as a starting point, right?

olikraus, 2022-09-21:
I have created a new constructor U8G2_SSD1306_96X39...

Can you test this beta release? Especially the u8g2 FlipMode test: There should be a solid frame at the display border all the time...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

fastfourier666, 2022-09-21:
OK! Thanks! I don't think I will be back at my bench until the weekend, but I will test it out then!

I used the 128X64 noname as a starting point. The 96X40 constructor resulted in "every other line". I tried messing with 0xDA and 0xC0/0xC8 but I couldn't ever get it to display solid.

olikraus, 2022-09-21:
Oh, you mean I should have better used the 128x64 init Code?

fastfourier666, 2022-09-21:
Yes, I was using the 128x64. I will try again with the 96X40 though. I might not get a chance to do it for a couple of days but I will certainly get back to you ASAP!

olikraus, 2022-09-21:
No issues, take your time...

fastfourier666, 2022-09-21:
OK! Plans changed and I had a bit of time to get back here and mess around with it. I'm using the `U8G2_SSD1306_96X39_F_HW_I2C` constructor from your beta release. Long story short, I can get the display to work properly! but `setFlipMode` and `setDisplayRotation` have a one pixel Y offset from each other.

The X axis is OK. with no sendF I get "every other line" in the Y axis. To get a full frame I had to do the two sendF commands in the below listing. A shortened version of your FlipMode sketch.

```
#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

U8G2_SSD1306_96X39_F_HW_I2C u8g2(U8G2_R0, 15);

void setup(void) {
  u8g2.begin();
  u8g2.sendF("ca", 0xDA, 0x10);       // cures "every other line"
  u8g2.sendF("ca", 0xD3, -1);         // cures Y offset 
  u8g2.setFont(u8g2_font_6x10_tr);
}

void draw(const char *s)
{
  u8g2.firstPage();
  do {
    u8g2.drawStr(2,15,"Hello World!");    
    u8g2.drawStr(2,30,s);    
    u8g2.drawFrame(0,0,u8g2.getDisplayWidth(),u8g2.getDisplayHeight() );
    u8g2.drawStr(2, 45, u8x8_u16toa(u8g2.getDisplayWidth(), 5));
    u8g2.drawStr(2, 60, u8x8_u16toa(u8g2.getDisplayHeight(), 5));
  } while ( u8g2.nextPage() );
  delay(2000);
}

void loop(void) {
  u8g2.clearDisplay();
  u8g2.setDisplayRotation(U8G2_R0);
  u8g2.setFlipMode(0);
  draw("R0,F0");
//  u8g2.clearDisplay();
//  u8g2.setFlipMode(1);
//  draw("R0,F1");

  u8g2.clearDisplay();
  u8g2.setDisplayRotation(U8G2_R1);
  u8g2.setFlipMode(0);
  draw("R1,F0");
//  u8g2.clearDisplay();
//  u8g2.setFlipMode(1);
//  draw("R1,F1");

  u8g2.clearDisplay();
  u8g2.setDisplayRotation(U8G2_R2);
  u8g2.setFlipMode(0);
  draw("R2,F0");
//  u8g2.clearDisplay();
//  u8g2.setFlipMode(1);
//  draw("R2,F1");

  u8g2.clearDisplay();
  u8g2.setDisplayRotation(U8G2_R3);
  u8g2.setFlipMode(0);
  draw("R3,F0");
//  u8g2.clearDisplay();
//  u8g2.setFlipMode(1);
//  draw("R3,F1");
}
```

With this sketch, `setDisplayRotation` rotates the display perfectly:

https://user-images.githubusercontent.com/33075175/191607044-cc0fe962-9f4e-48b6-8a20-e3c4bf15bfe2.mp4

If I uncomment the `setFlipMode` commands from the sketch, they give me a Y offset of 1:

https://user-images.githubusercontent.com/33075175/191607664-bc14f7ee-9082-43b9-9a8d-c5f6bf6f982d.mp4

If I use that second sendF command to set the Y offset to 0 instead of -1,  the `setFlipMode` commands rotate the display properly but the `setDisplayRotation` gives me an offset of 1 again.

I guess in my application, `setFlipMode` would be the better option, since the hardware supports it?

Also. One more question. Is it possible to have a 4-wire SPI version of this constructor? I am driving 12 displays with this project, and turns out that I2C is too slow. At least I could not get it to go above 800kHz on an ESP32.

Thanks so much for your time!

olikraus, 2022-09-25:
> I guess in my application, setFlipMode would be the better option, since the hardware supports it?

yes, I will look into it.

> Also. One more question. Is it possible to have a 4-wire SPI version of this constructor? 

It should be included already. All these constructors are automatically created. It should be listed in the .h file and called 4W_HW_SPI or so. Let me know if you don't find it.

olikraus, 2022-09-25:
I have created a new beta 2.34.2. Thanks for the videos, however I am still a little bit lost of what all is required and whether it new works?
Can you run the unmodified flipmode example with all 8 states and send me the result here if it still doesn't work?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

fastfourier666, 2022-09-25:
I knew I could have explained that better!

The previous beta you posted is perfectly usable for me in its current state using FlipMode only.

I was just posting those to illustrate that I could get _either_ `setFlipMode` OR `setDisplayRotation` to work, but not both at the same time without doing some `sendF`. For other people that might use this display.

To be honest I'm not 100% sure what the difference is between flipMode and setDisplayRotation, but I am very thankful that you did this!

I will test in the next couple of days once I am back home.

Thank you!

olikraus, 2022-09-25:
> The previous beta you posted is perfectly usable for me in its current state using FlipMode only.

We can keep everything as it is at the moment, however it would be great if `setFlipMode` and `setDisplayRotation` would work.

> To be honest I'm not 100% sure what the difference is between flipMode and setDisplayRotation, but I am very thankful that you did this!

Both are kind of redundant, but flipMode uses a hardware feature of the display (if available). Display rotation can be used if the display hardware doesn't support 180 degree flipping or if 90 degree or mirror display mode are required.
(Mirror mode had been requested by some funny DIY projects which use a real physical mirror.)

Ideally both should work in parallel

fastfourier666, 2022-09-25:
Got back and had a chance to test. Here is a video of the unaltered FlipMode sketch running:

https://user-images.githubusercontent.com/33075175/192170478-06bf3711-8442-472e-a9a4-7078a7f7f5f0.mp4

Looks like there is a way of getting it to display correctly in all orientations, but you need to use a combination of flipmode and rotation to get it done.

This will work great for me as-is, but I'm of course happy to do any more testing if you want to send any more beta builds to make those functions work the same as for the other constructors.

olikraus, 2022-09-29:
Thanks for the video. Let's do a final attempt. I have modified the y delta for F0 by 1.

I have created beta 2.34.3...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

fastfourier666, 2022-10-06:
Hi Oli, I had planned to get to this this week but my only 96X39 OLED met with an accident. I dropped a screwdriver on it and cracked it. Currently waiting for some more :(

miraclehoshi, 2022-10-20:
https://user-images.githubusercontent.com/35333519/196889287-97fc34ef-df78-4559-9586-40233c9cf31e.mp4

hello! I recently came across this issue while testing by ordering the same display (different colors).

When using the flip mode example using the 2.34.4 version of the library, it works as above. I think the y-axis should go down 1 pixel..... I hope this helps.

olikraus, 2022-10-20:
Which constructor did you apply?

miraclehoshi, 2022-10-21:
> Which constructor did you apply?

U8G2_SSD1306_96X39_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

I didn't do any other settings. This is the result of uncommenting that line in the example and running it.

olikraus, 2022-10-22:
After u8g2.begin(): If you do a
``` C++
u8g2.sendF("c", 0x41);
```
How will the same example look like?

miraclehoshi, 2022-10-23:
```
void setup(void) {
  u8g2.begin();
  u8g2.sendF("c", 0x41);
  u8g2.setFont(u8g2_font_6x10_tr);
}
```

https://user-images.githubusercontent.com/35333519/197373116-578cb172-7b29-4c46-be97-a801a4333e9e.mp4

olikraus, 2022-10-23:
Strange, somehow there seems to be no difference between the two videos. Flip mode 0 is wrong in both cases.
Will there any change if you increase 0x41 to higher values? Maybe like 0x48 or even higher?

miraclehoshi, 2022-10-25:
It didn't respond to values including 0x40.
 It's definitely weird. It was the same even if I looked at the datasheet and modified it according to the reference.

I think you have the option to either send that display or provide a simple RDP environment where the display is connected with an arduino that can be accessed remotely. Which one looks better?

olikraus, 2022-10-25:
I don't understand?

miraclehoshi, 2022-10-25:
I would like to send a 96x39 display to you by this courier.

Or I could connect this OLED with an Arduino, connect it to a remotely accessible computer, 
watch a webcam, and work as if you had a real device.

Which one do you think would be more comfortable? I love this 96x39 display, it can propel anything.

Of course, at the level where it is not rude. If it bothers you or you don't have time, you can say no.
I can't understand to the level of my knowledge why this display doesn't work as per the datasheet.

olikraus, 2022-10-25:
Ah , now I got your points. Well no issues, I can order.

Is this the device?
https://de.aliexpress.com/item/1005004166150651.html

miraclehoshi, 2022-10-25:
Yes, that's right. One order over there is what I ordered.

miraclehoshi, 2022-10-25:
Alright, I've spent the whole day trying to figure out the cause.

1. Failed to follow datasheet.
The oled manufacturer cut off the top line instead of cutting the bottom line on the 96*40 display. I don't know why they did this, but this display is set to row1-40, not row0-39.
[image]
`  U8X8_CA(0x0d3, 0x03F)`
In other words, to move down one space, we were able to solve it by giving the display offset 63 spaces (64-1 spaces).

2. Abnormal behavior of u8g2
I noticed some strange things while debugging. In the flip demo conducted by me and the fastfourier666, strange pixels are displayed at the bottom.
[image]
This symptom occurred in all u8g2 examples. It didn't happen in the u8x8 example.
I used the u8g2 to light up all the pixels and found that only the 35 pixels above were illuminated. In the 96*4 area below, only garbage values were displayed.
[image]
Here you can see that something is wrong with the u8g2's display area settings. The 4 pixels below are garbage values displayed out of bounds. When I connected multiple displays at the same time, I knew what was displayed below was different and I had a gut feeling that it was a garbage value.
In other words, it boils down to the following two problems.
1. The display starts at 1, not row0. You need to give it an offset of 63 pixels.
2. Something is wrong with u8g2's settings. This is why the offset instruction did not work. but I couldn't analyze this because I didn't know the details.

olikraus, 2022-10-25:
Thanks for the detailed analysis. And yes, we need to teach u8g2 regarding the architecture. 
I will order one of those parts.

miraclehoshi, 2022-10-25:
I think the easiest way to solve the u8g2 problem is to just use a code like 96x40 and give it an offset of 63.
It's the same as a 96x40 display with the top line cut off.

olikraus, 2022-11-01:
> I think the easiest way to solve the u8g2 problem is to just use a code like 96x40 and give it an offset of 63. It's the same as a 96x40 display with the top line cut off.

This is what I did, but it seems to be more complicated.

I have ordered one of those devices.

olikraus, 2022-12-02:
Today the display arrived. I hope to have time to look into this in the next days.

olikraus, 2022-12-04:
It should be fixed now. 
I have created beta 2.34.6
You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

miraclehoshi, 2022-12-11:
I tested several examples. No particular problems were found in the test.
Everything seems to work just fine! Can you tell me what could be the cause of the problem?

olikraus, 2022-12-11:
I fixed the start line.

miraclehoshi, 2022-12-12:
ah.

fastfourier666, 2022-12-14:
Hi Oli,
Very sorry about my absence from this thread! It was unavoidable unfortunately :/
Thank you again for your work on this. I am still waiting for my displays (where have they gone?!) but it looks like this will work great for everyone! Appreciate you!

closed 2022-12-14 by fastfourier666

olikraus, 2022-12-14:
No problem at all. There's no pressure to close issues here 😃