olikraus/u8g2 #1960: ZJY-M150-128X128 SSD1327 OLED

cranefist, 2022-09-18:
This OLED module:
https://www.aliexpress.com/item/1005003784174529.html

Driver : ssd1327
Model : zjy-m150-128x128

The SPI model works with : U8G2_SSD1327_MIDAS_128X128_F_4W_HW_SPI
Does not seem to need the RESET pin.

And I2C : U8G2_SSD1327_MIDAS_128X128_F_HW_I2C

But it needs an extra command that enables the voltage converter : u8g2.sendF("ca", 0xB5, 0x03); 

If you want to add it, seen few post about these displays and people having problems using them and thinking they are broken.

olikraus, 2022-09-18:
I have created a new beta version with this extra command. Can you test this version?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

cranefist, 2022-09-18:
Works fine! Tested on SPI. I think i have the I2C version also somewhere, but i expect that it would work also.

olikraus, 2022-09-19:
😁 thanks for the feedback

closed 2022-09-26 by cranefist