olikraus/u8g2 #1961: Winstar WF62ATXGRDNN0# with ST7511U driver

talinham, 2022-09-18:
Hello,

I would like to know if I can use the existing library for ST7511 for the ST7511U? I have a Winstar TFT LCD 640*320 dot matrix and it supports parallel 8080 and 6800 and SPI 3 and 4 lines. I need some guide to know what to modify to make the library compatible with the LCD I got.
[WF62ATXGRDNN0#.pdf](https://github.com/olikraus/u8g2/files/9595121/WF62ATXGRDNN0.pdf)
[ST7511U-Sitronix.pdf](https://github.com/olikraus/u8g2/files/9595122/ST7511U-Sitronix.pdf)

olikraus, 2022-09-18:
You could start with the existing constructor: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7511-avd_320x240

Will it show something? Maybe you need to change the contrast value for this display.
If you can see something, then a picture would be great.

Ref: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7511.c

talinham, 2022-09-19:
I've used the following constructor and tries to run the hello world 8x8 example. 
U8X8_ST7511_AVD_320X240_3W_SW_SPI u8x8(/*clock=*/12, /*data=*/13,/*cs=*/ 10 , /*reset=*/ 8);

so far nothing shows up. I need some help with what to modify to make the library compatible with the lcd. 
I connected the lcd directly to the Arduino Mega Pro. I am not sure if I need to define anything for the pins? 
I would appreciate if you can give me some pointers.

olikraus, 2022-09-19:
> I would appreciate if you can give me some pointers.

I suggest to use the 4-line serial interface as described in section 7.3
Line A0 is called DC in u8g2.
Both IF inputs need to be high for 4-wire SPI. Did you also follow all the pin assignments, mentioned in section 9.1 of your display datasheet?
Display : u8g2
D7: Clock
D6: Data
A0: DC
/CS: CS

Connect /WR and /RD as adviced by the datasheet.
My suggestion is either not to enable the backlight LED or to use an external power supply( Do NOT use the power supply from your Arduino board for the backlight LED)
Add a 100nF cap close to the power supply lines of the flex connector of your display.

talinham, 2022-09-19:
I am not sure if I'm doing my wiring right. I'm using Arduino Mega Pro and the pin assignment is different from Arduino. 
LCD       Mega Pro
D7           21 (SCL)
D0           20  (SDA)
A0            9
CS            10
[image]

not sure if this is correct?

talinham, 2022-09-19:
I am not sure if I'm doing my wiring right. I connected R/W, RD(e) and DB1 to DB6 to high. I'm not sure how to connect the rest so it's correct with the pin configuration of Arduino. I'm using Arduino Mega Pro and the pin assignments are a bit different. 
This is the pinout for it:
[image]
I used the below pinout: 
DB0(SDA) to pin D51(MOSI)
DB7(SCL) to pin D52(SCK)
CS to pin D53(SS)
A0 to pin D50(MISO)

olikraus, 2022-09-20:
Hmm.. Atmel SPI has some issues with hardware SPI. U8g2 will not use hardware controlled CS but instead does it manually. However if I remember corretly Atmel SPI will disallow manual usage of CS if hardware SPI is enable (same is probably true for MISO). As a consequence: You should better use other pins for CS and A0 (and leave D53 and D50 unconnected).

Just for testing you could use software emulated SPI with u8g2. If HW SPI is not used, then pins 53 and 50 are usable:
```
U8X8_ST7511_AVD_320X240_4W_SW_SPI( /* clock */ 52, /* data */ 51, /* cs */ 53, /* dc */ 50, /* reset */ ... )
```

I suggest also to use the reset input of the display and connect it to a GPIO port of your controller.

Please consider to use the contrast test example from u8g2, because the contrast setup might be different with your display.

closed 2023-02-22 by olikraus