olikraus/u8g2 #1965: Please help reduce font size u8g2_font_unifont_t_vietnamese2 to 5*7 or 6*8 to display properly on 8*8 Max7219 matrix.

nphumy, 2022-09-21:
Dear Mr Olikraus. 
I really like your library because it's easy to use and available in many languages. I am Vietnamese; and I want to use Vietnamese to display the matrix 8*8 Max 7219. I have used 8 modules 8*8 Max7219 ; concatenate in a horizontal row and use the U8g2 library; and used the font  "u8g2_font_unifont_t_vietnamese2 " of your Library to display Vietnamese. When I upload sketch in Arduino IDE The screen displays the correct text but the height of the characters is only 1/2 . Maybe this font is designed in 8*16 font. Can you please help me to reduce the font size of font " u8g2_font_unifont_t_vietnamese2"  to 5*7 or 6*8 (or 8*8) size  so that the characters can be displayed properly on 8 lines of the Max7219 matrix?
Thank you very much for your help.

olikraus, 2022-09-21:
U8g2 fonts are bitmap fonts, they can't be scaled. However you can provide a suitable bitmap font which can be converted into u8g2 format.

nphumy, 2022-09-27:
Thank you for your very quick response. I have already converted the Vietnamese font.

closed 2022-09-27 by nphumy