olikraus/u8g2 #1970: GU800 160X16

chicho70, 2022-10-03:
Hi,

I have a Noritake GU800-160X16.

In U8X8 i can write two lines of 20 chars each. No problems. Only 1X1 and 2X2 fonts.

But with U8G2 a can only use 128 bits (trunked to 16 chars). I only can write in the first 128 pixels (from 129 to 160 are blank)

In u8x8_d_gu800.c file I did some modifications:
		
/* sck_clock_hz = */ 8000000UL,  	   <--
/* spi_mode = */ 3,            	                  <-- 
/* i2c_bus_clock_100kHz = */ 4,           /* GU800: Not used */
/* data_setup_time_ns = */ 40,    
/* write_pulse_width_ns = */ 150, 
  /* tile_width = */ 16,                             <--
  /* tile_hight = */ 8,             
  /* default_x_offset = */ 0,       
  /* flipmode_x_offset = */ 0,      
  /* pixel_width = */ 160,                      <--
  /* pixel_height = */ 16                       <--
};

Thanks and best regards.

olikraus, 2022-10-03:
Shell we create a new constructor with your modifications?

chicho70, 2022-10-03:
yes, i changed SPI from 2 to 3. I found information about it.
With SPI =2  not ok.

i could be fine a new constructor for NORITAKE GU800 160X16.

chicho70, 2022-10-04:
U8X8 Results: Full Screen
/Users/fausto/Desktop/carpeta sin título/pic1.jpg

chicho70, 2022-10-04:
U8G2 Results: Max to 128 pixels and overlapped.
[image]

chicho70, 2022-10-04:
I leave more info for you:
[Noritake Display GU160X16-800A.pdf](https://github.com/olikraus/u8g2/files/9706282/Noritake.Display.GU160X16-800A.pdf)
[GU128X32-800B.pdf](https://github.com/olikraus/u8g2/files/9706283/GU128X32-800B.pdf)

chicho70, 2022-10-06:
Thanks so much.

olikraus, 2022-10-09:
I have created a new constructor GU800_160X16.
Please have a look.
You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```
#

chicho70, 2022-10-14:
Hi Oliver,
Works so pretty.
Thanks so much.
Best regards.

Noritake GU160X16-800A display mounted in GPS TRUETIME / Symmetricom MODEL XLi TIME & FRECUENCY SYSTEM

[image]

chicho70, 2022-10-14:
Oliver, 
When you have time, please, if you can, take a look at this display.
I have tried to adapt the original Noritake library developed for AVT to ESP32 but it is impossible for me.
The original library is supported by these three files:

#include <util/delay_basic.h>
#include <util/delay.h>
#include <avr/pgmspace.h>

I leave you the original library in case it helps you for this display (GU256X32D-3900B).
I also send you the PDF file of the display information to help you.

It is possible that one of the constructors in your u8g2 project can fit this display.

This Noritake display, model GU256X32D-3900B, was installed on a GPS Symmetricom MODEL XLi TIME & FREQUENCY SYSTEM and with the GU160X16 I want to make a small weather station using the Sensirion SCD41 sensor.

Once again, thank you very much.

[image]

[image]

olikraus, 2022-10-14:
Thanks for the feedback.

chicho70, 2022-10-27:
Dear Oliver,
I´ve checked with this constructor with GU160X16-800A (Paralell M68/68000) & ESP32 DEVKIT V1 ... and works fine. 
Thanks and regards. 

U8G2_GU800_160X16_F_6800 u8g2(U8G2_R0, /*d0*/4, /*d1*/16, /*d2*/17, /*d3*/18, /*d4*/19, /*d5*/21, /*d6*/22, /*d7*/23, /*ENABLE-ENCK*/27, /*CS-CSS*/14, /*DC-C/D*/26, /*reset*/13);

REMARKS: PIN 17 (R/W)  GU160X16 to GND to force WRITE.

[image]

olikraus, 2022-10-27:
Thanks for the feedback!

closed 2023-02-22 by olikraus