olikraus/u8g2 #1974: Noritake GU256X32D-3900B Help

chicho70, 2022-10-05:
Hi,
Do you have any code for this Noritake GU256X32D-3900B display for Arduino (ESP32) ?.
This display has 2 interfaces: RS232 and CMOS parallel.
The code for both will be better, although for RS232 will be enough.
Thanks and best regards.

[image]

[image]

olikraus, 2022-10-06:
> Do you have any code for this Noritake GU256X32D-3900B display for Arduino (ESP32) ?.

I don't think so...

chicho70, 2022-10-06:
Regards.

chicho70, 2022-10-14:
Dear Oliver,
when you have time, please, if you can, take a look at this display.
I have tried to adapt the original Noritake library developed for AVT to ESP32 but it is impossible.
The original library is supported by these three files:

#include <util/delay_basic.h>
#include <util/delay.h>
#include <avr/pgmspace.h>

I leave you the original library in case it helps you for this display (GU256X32D-3900B).

It is possible that one of the constructors in your u8g2 project can fit this display.

I also send you the PDF file of the display information.

This Noritake display, model GU256X32D-3900B, was installed on a GPS Symmetricom MODEL XLi TIME & FREQUENCY SYSTEM and with the GU160X16 I want to make a small weather station using the Sensirion SCD41 sensor.

Once again, thank you very much.

[Arduino_Noritake_VFD_GU7000_V2.zip](https://github.com/olikraus/u8g2/files/9787547/Arduino_Noritake_VFD_GU7000_V2.zip)
[Arduino_Noritake_VFD_GUD900.zip](https://github.com/olikraus/u8g2/files/9787548/Arduino_Noritake_VFD_GUD900.zip)
[GU256X32 PDFs Info.zip](https://github.com/olikraus/u8g2/files/9787555/GU256X32.PDFs.Info.zip)

olikraus, 2022-10-14:
The implementation for the GU800 had been very difficult: https://github.com/olikraus/u8g2/issues/1604

So I actually hesitate to do this again. If you would use one the existing GU800 constructor with the above mentioned displays, will you see something?

chicho70, 2022-10-15:
Hi Oliver,
I really appreciate the effort you've put into developing the code for the GU800.
The GU800's screen phosphor is worn and has less resolution than the GU256X32.
Thank you so much for everything.

olikraus, 2022-10-17:
What will you see on the GU256x32 if you use the U8G2_GU800_128X64 constructor? Obviously the result can not be correct, but still, what will be visible?

chicho70, 2022-10-17:
On GU800 i am using SPI.
And on GU256X32 with U8G2_GU800_128X64 constructor, i have to connect using parallel interface.
I,ve never used parallel interface, but i,ll try following your suggestion. Then i'll keep you informed.
Regards.
Gracias.

chicho70, 2022-10-18:
Dear Oliver,
In your last release i can see only these  U8G2 contructors for GU800, all of them for SPI interface.

The GU256X32D has two interfaces: Serial interface (Asynchronous, RS-232 -level) and Parallel interface (CMOS), so i can´t check U8G2_GU800_128X64 constructor for it.

U8G2_GU800_128X64_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_GU800_128X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_GU800_160X16_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_GU800_160X16_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

Best regards.
Muchas gracias.

chicho70, 2022-10-18:
Sorry, I realised that in your **U8g2lib.h** file i´ve found these constructors:

U8G2_GU800_128X64_F_6800(const u8g2_cb_t *rotation, uint8_t d0, uint8_t d1, uint8_t d2, uint8_t d3, uint8_t d4, uint8_t d5, uint8_t d6, uint8_t d7, uint8_t enable, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE)

U8G2_GU800_128X64_F_8080(const u8g2_cb_t *rotation, uint8_t d0, uint8_t d1, uint8_t d2, uint8_t d3, uint8_t d4, uint8_t d5, uint8_t d6, uint8_t d7, uint8_t enable, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE)
 
U8G2_GU800_160X16_F_6800(const u8g2_cb_t *rotation, uint8_t d0, uint8_t d1, uint8_t d2, uint8_t d3, uint8_t d4, uint8_t d5, uint8_t d6, uint8_t d7, uint8_t enable, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE)

U8G2_GU800_160X16_F_8080(const u8g2_cb_t *rotation, uint8_t d0, uint8_t d1, uint8_t d2, uint8_t d3, uint8_t d4, uint8_t d5, uint8_t d6, uint8_t d7, uint8_t enable, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE)

According datasheet, G256X32D display use D0~D7 and /WR signals. 

[image]

In my case, which signal should i use?, only ENABLE signal  to WRITE ?

chicho70, 2022-10-28:
Dear Oliver,
I´ve tried with these constructors and same bad results:

U8G2_GU800_160X16_F_6800 u8g2(U8G2_R0, /*d0*/4, /*d1*/16, /*d2*/17, /*d3*/18, /*d4*/19, /*d5*/21, /*d6*/22, /*d7*/23, /*ENABLE-WRITE*/13, /*CS-CSS: al aire N/C*/26, /*DC-C/D: al aire N/C*/27);

 U8G2_GU800_160X16_F_8080 u8g2(U8G2_R0, /*d0*/4, /*d1*/16, /*d2*/17, /*d3*/18, /*d4*/19, /*d5*/21, /*d6*/22, /*d7*/23, /*ENABLE-WRITE*/13, /*CS-CSS: al aire N/C*/26, /*DC-C/D: al aire N/C*/27);

U8G2_GU800_128X64_F_6800 u8g2(U8G2_R0, /*d0*/4, /*d1*/16, /*d2*/17, /*d3*/18, /*d4*/19, /*d5*/21, /*d6*/22, /*d7*/23, /*ENABLE-WRITE*/13, /*CS-CSS: al aire N/C*/26, /*DC-C/D: al aire N/C*/27);

U8G2_GU800_128X64_F_8080 u8g2(U8G2_R0, /*d0*/4, /*d1*/16, /*d2*/17, /*d3*/18, /*d4*/19, /*d5*/21, /*d6*/22, /*d7*/23, /*ENABLE-WRITE*/13, /*CS-CSS: al aire N/C*/26, /*DC-C/D: al aire N/C*/27);

// End of constructor list

void setup(void)
  {
    u8g2.begin();
    u8g2.setContrast(0);
  }

void loop(void)
  {
    u8g2.clearBuffer();					// clear the internal memory
    u8g2.setFont(u8g2_font_8bitclassic_te); // bonita
    u8g2.drawStr(0,12,"Hello World 12378094581");	// write something to the internal memory    
    u8g2.sendBuffer();					// transfer internal memory to the display
    delay(5000);  
  }

[image]

[image]

I can help you to test what you need.

Regards

olikraus, 2022-10-28:
To me it looks like your display is a character display. I probably can not do anything here...

chicho70, 2022-10-28:
Maybe, I'll read the documentation in case I can do something.
I appreciate all the effort you have put into both GU160X16 and this one.
Thank you very much Oliver.

bsehovac, 2025-01-15:
@chicho Have you ever managed to make this display work? Have you tried the [Noritake Arduino library](https://www.noritake-elec.com/support/design-resources/support-guide/arduino-code-library-quick-start-guide) for it?   BTW, do you know where I can buy this display?

chicho70, 2025-01-15:
Hi,

Yes, It was installed in an old Simmentricom GPS.

Now i´m on holidays. I´ll give more information when I return to work.

For now, I send you two project2 for ESP32, both in wrote in Spanish but easy to understand, and PDF information that I  have collected about this display.

I remember that I tried to use another different fonts as you developed for GU160X16-800A but for me I was very difficult to develop.

I hope to help you amigo.

I send in differents mails …

Fecha: miércoles, 15 de enero de 2025, 10:34
Para: olikraus/u8g2 ***@***.***>
CC: chicho70 ***@***.***>, Author ***@***.***>

chicho70, 2025-01-15:
Connection ESP32 - Display
***@***.***

chicho70, 2025-01-15:
Problems to send you PDF  files.

Part 1

chicho70, 2025-01-15:
..  Part 2

chicho70, 2025-01-15:
… and part 3.

Regards  😉