olikraus/u8g2 #1977: Completely Random Visual Artifacts & Freezing w/ SSD1322 + ATmega32U4

SnoWFLakE0s, 2022-10-06:
Hello, 

First, thank you for the library. It is incredibly helpful.

I have purchased one of the 256x64 OLEDs using a SSD1322 driver and have it connected to a Pro Micro (32U4).

The display worked fine with this library and I have footage of it functioning just fine mere days ago. As I was polishing up my program, I tested the display again today and I am completely unable to get it to work properly.

There are two primary issues:

One, the display freezes and begins to produce a fuzzy square in some corner of the display, or simply does not display, after some time displaying images (it previously did not). I have attached footage of this phenomenon. I loaded the page buffer version of the GraphicsTest onto the uC for this.

https://user-images.githubusercontent.com/32413097/194243656-6a680c33-5bce-4a2d-9690-301c139584a2.mp4

Second, I have experienced issues with the display somehow failing to initialize & begin displaying, independent of code somehow. By simply unplugging and re-plugging the USB cable to the uC, it usually fixes itself. The length at which the display can proceed without issue seems to be completely random each time I re-power the uC.

The following is a photo of my wiring.

[image]

Is there anything I can do to fix these issues? I would really appreciate some help.

SnoWFLakE0s, 2022-10-06:
To add, not all examples seem to function. The end goal of my application is essentially a serial monitor, and just yesterday the Serial example was working fine... Today I am getting no display output at all.

olikraus, 2022-10-06:
1. Did you enable 16 bit mode for u8g2? it is required in your case due to the width of your display (see FAQ)
 2. It is in general not a good idea to use these protoboards. Better solder everything on a proto PCB
 3. Using resistors for level translation is also not a good and reliable approach. Better use a proper approach like a 74HC4050.

SnoWFLakE0s, 2022-10-06:
> 1. Did you enable 16 bit mode for u8g2? it is required in your case due to the width of your display (see FAQ)
> 2. It is in general not a good idea to use these protoboards. Better solder everything [...]

1. As far as I understand, yes.
[image]
2. Good point. I doubt it is an electrical connection fault for the most part however, as it does produce a proper output... for a limited time, that is.
3. I will try that.

Outside of these points, is there any reason why this is suddenly dysfunctional? It was just fine quite literally yesterday.

SnoWFLakE0s, 2022-10-07:
New jumpers, tested for continuity. No problems.

Still no sauce either on the ESP32 or the Pro Micro. I get completely zero signs of life from the display now. Is the display dead?
I am not sure how I can confirm that, but for the sake of testing I reversed the CS and DC pins and got something to show.

[image]

Would this imply that the display is still alive?

olikraus, 2022-10-09:
Last picture looks actually good.

SnoWFLakE0s, 2022-10-09:
> Last picture looks actually good.

Any pointers on why no output otherwise? I can't figure it out, for the life of me...

olikraus, 2022-10-09:
Which software did you use to produce the output on the display?

SnoWFLakE0s, 2022-10-09:
> Which software did you use to produce the output on the display?

Been using page buffered GraphicsTest.ino for my troubleshooting. Serial.ino produces no output.

olikraus, 2022-10-09:
Which constructor?

SnoWFLakE0s, 2022-10-09:
> Which constructor?

`U8G2_SSD1322_NHD_256X64_1_4W_HW_SPI`.
I also tried the SW SPI version, not to mention text-only 8 bit. Neither function.

olikraus, 2023-02-22:
Looks like I missed your response here. Did you solve your problems? What is the current status?

hipur, 2023-08-24:
I'm facing the same issue with ESP32 boards. How did you solve this problem?