olikraus/u8g2 #1978: LCD st7920 128x64 not showing

celsoluiz81, 2022-10-08:
Hello,

First of all thank for developing this. I'm sure it will be of great use as soon as I am able to get it running.

I have an issue running a simple "Hello World" example on my ST7920 128x64 LCD Display. I am using a small ESP32 development board. I'm developing in C using Espressif's Framework (in other words, not Arduino IDE).

When I flash the code to the controller and hit "EN", the LCD flickers for a second, but nothing is ever shown on the screen. My expectation is to see the words "Hello World!".

Also, when I run the **idf.py monitor** on the command line, I see the Log text "Sent buffer" every 10 ms, which is placed in my WorkstationLcdMain function just after calling SendBuffer().

I have checked that the pins are correctly connected to the LCD and ESP32.  On the LCD screen it appears that all pixels are "filled" (See screenshot).

Any help or tip here will be very much appreciated! 

I'm using nKolban's library to communicate with my display via SPI.
https://github.com/nkolban/esp32-snippets/tree/master/hardware/displays/tests 

My pin configuration is as follows:

**workstation_cfg.h**
```
/* LCD Pins */
#define WORKSTATION_LCD_PIN_RS        GPIO_NUM_5 /* CS */
#define WORKSTATION_LCD_PIN_RW        GPIO_NUM_23 /* MOSI */
#define WORKSTATION_LCD_PIN_E         GPIO_NUM_18 /* CLK */
#define WORKSTATION_LCD_PIN_RST       GPIO_NUM_19

```

Here are my code snippets:

**workstation_lcd.c**
```

#include <stdio.h>
#include <string.h>
#include "driver/gpio.h"
#include "workstation_lcd.h"
#include "driver/spi_master.h"
#include "esp_log.h"
#include "sdkconfig.h"

#include "../../main/workstation_cfg.h"
#include "../u8g2/csrc/u8g2.h"
#include "../u8g2/csrc/u8g2_esp32_hal.h"

static const char *TAG = "LCD";
u8g2_t u8g2; // a structure which will contain all the data for one display

void WorkstationLcdInit(void){

  u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;

	u8g2_esp32_hal.clk   = WORKSTATION_LCD_PIN_E;
	u8g2_esp32_hal.mosi  = WORKSTATION_LCD_PIN_RW;
	u8g2_esp32_hal.cs    = WORKSTATION_LCD_PIN_RS;
	u8g2_esp32_hal.reset = WORKSTATION_LCD_PIN_RST;
	u8g2_esp32_hal_init(u8g2_esp32_hal);

  u8g2_Setup_st7920_128x64_1(
		&u8g2,
		U8G2_R0,
		u8g2_esp32_spi_byte_cb,
		u8g2_esp32_gpio_and_delay_cb);  // init u8g2 structure

  ESP_LOGI(TAG, "Initializing Display");
  u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,

  u8g2_SetPowerSave(&u8g2, 0); // wake up display

  ESP_LOGI(TAG, "Initializing LCD Task");
  xTaskCreate(lcd_task, "lcd_task", 4096, NULL, 10, NULL);
}

void WorkstationLcdMain(void){
  u8g2_ClearBuffer(&u8g2);

  u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
	u8g2_DrawStr(&u8g2, 0,15,"Hello World!");
  u8g2_SendBuffer(&u8g2);
  ESP_LOGI(TAG, "Sent buffer");
}

static void lcd_task(void* arg){
  while (1) {
    WorkstationLcdMain();
    vTaskDelay(10);
  }
}
```
[image]

olikraus, 2022-10-08:
I can not say anything regarding none-Arduino environments, but maybe it is also a hardware issue in your case: Did you add a varpot? The varpot is at least not visible an your picture, but there must be one.

celsoluiz81, 2022-10-16:
Hello. Thanks for your prompt reply. I do have a 10k potentiometer wired according to this tutorial:
https://www.instructables.com/ST7920-128X64-LCD-Display-to-ESP32/

olikraus, 2023-02-22:
so i assume it works now...

closed 2023-02-22 by olikraus