olikraus/u8g2 #1984: help to find lcd driver model :)

mohammad0663, 2022-10-14:
hi olikraus .
i have a graphic lcd 12864 (monochorom) . i do finde pin out and hex whith logic analayzer . but can't finde driver !
at down , i did write otp (i did finde with logic analayzer) :

  LcdWrite(LCD_C, 0x28);
  LcdWrite(LCD_C, 0x23);
  LcdWrite(LCD_C, 0xa0); 
  
  
  LcdWrite(LCD_C, 0x2f);
  LcdWrite(LCD_C, 0x81);
  LcdWrite(LCD_C, 0x4b);
  
  
  LcdWrite(LCD_C, 0x24);
  LcdWrite(LCD_C, 0xeb);
  LcdWrite(LCD_C, 0xc4);
  
  
  LcdWrite(LCD_C, 0xb0);
  LcdWrite(LCD_C, 0x10);
  LcdWrite(LCD_C, 0x04);
  LcdWrite(LCD_C, 0x40);
  
  
  LcdWrite(LCD_C, 0xa4);
  LcdWrite(LCD_C, 0xa6);
  LcdWrite(LCD_C, 0x88);
  
  
  LcdWrite(LCD_C, 0xaf);
  
  
  LcdWrite(LCD_C, 0xb0);
  LcdWrite(LCD_C, 0x10);
  LcdWrite(LCD_C, 0x04);

What driver model is this otp For ?

What driver can I use?

olikraus, 2022-10-15:
I don't know how I can help here.

mohammad0663, 2022-10-15:
Tell me whte dirver number ?

olikraus, 2022-10-15:
Impossible without datasheet. But it could be a ST7565.

mohammad0663, 2022-10-16:
I do test . Thank you

closed 2023-02-22 by olikraus

mohammad0663, 2023-02-22:
Hello, I tested it and it was uc1601