olikraus/u8g2 #1988: Using defined constants in MUI_FORM

kasitoru, 2022-10-18:
I would like to be able to use defined constants as arguments for `MUI_FORM` macro:

```c
#define MENU_FORM 1
#define MAIN_FORM 2
....
static const fds_t fds_data[] MUI_PROGMEM =
    MUI_FORM(MENU_FORM)
    ....
    MUI_FORM(MAIN_FORM)
;
....
```
It is impossible to do this now:
```
u8g2/csrc/mui.h:547:25: error: expected ',' or ';' before 'MUI_MENU_FORM'
  547 | #define MUI_FORM(n) "U" MUI_##n
      |                         ^~~~
../main.c:99:9: note: in expansion of macro 'MUI_FORM'
   99 |         MUI_FORM(MENU_FORM)
```
As a hack, I use an additional level of macro definition:
```c
#define MUI_FORM2(n) MUI_FORM(n)
```
But I would like it to work by default. This feature would increase the readability of the code.

olikraus, 2022-10-18:
Will this work?
``` C++
#define MUI_MAIN_FORM MUI_FORM(1)
```

Using constants is probably not possible (well i might need to check ).

kasitoru, 2022-10-18:
> Will this work?

It's inconvenient. Constants, for example, can also be used with other functions:
```c
mui_GotoForm(&mui, MAIN_FORM, 0);
```
In any case, I found a way out for myself. I'm just informing you that this doesn't work by default.