olikraus/u8g2 #1989: Cannot get SH1122 to display (Tests with ESP32, Arduino Uno, ATmega32u4)

SnoWFLakE0s, 2022-10-20:
Hello,

After my disappointing experience with the SSD1322 256x64 OLEDs I decided by then try a similar but smaller display--2.08" SH1122 OLEDs. I ordered two pieces in case one fails.

**Neither function.** I have made sure that 16-bit mode is enabled in u8g2.h, the pins are mapped correctly, and that the voltages going into the display's pins are 3v3 nominal. 

In these examples I am trying the GraphicsTest example. I have also attempted the HelloWorld example, which similar fails.

On the AVR platforms I simply do not get a display out at all. See below:
[image]

On the ESP32 I get *something*, but nothing that it should display. Just a white noisy screen.

https://user-images.githubusercontent.com/32413097/196875562-45058087-26b2-4bb6-8df1-81042b1bd359.mp4

What am I doing wrong? I would really appreciate some help.

olikraus, 2022-10-20:
In general those protoboards are not that reliable, but well...
Which constructor did you use?

SnoWFLakE0s, 2022-10-20:
> In general those protoboards are not that reliable, but well... Which constructor did you use?

`U8G2_SH1122_256X64_1_4W_SW_SPI`.
I verified proper continuity of the pins when I did not get an output. I also tried directly connecting to the ESP32 board and got the same result, so not an issue of the breadboard.

olikraus, 2022-10-21:
Did you use one of the existing examples from u8g2? Or: Which exact code did you use for testing?

SnoWFLakE0s, 2022-10-21:
> Did you use one of the existing examples from u8g2? Or: Which exact code did you use for testing?

I changed nothing minus removing the comment-out for the proper constructor. Using the bundled page buffer examples.

olikraus, 2023-02-22:
Probably you might need to update the pins. All in all information is little bit less to support...

GancFlex, 2023-05-18:
i have same problem

olikraus, 2023-05-18:
without further information I am not able to say anything. What is the problem? How does your constructor look like? How is your display connected to which board?

GancFlex, 2023-05-18:
> without further information I am not able to say anything. What is the problem? How does your constructor look like? How is your display connected to which board?

im using arduino nano okay and im using U8G2_SH1122_256X64_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);	

Display does't show anything

GancFlex, 2023-05-18:
> without further information I am not able to say anything. What is the problem? How does your constructor look like? How is your display connected to which board?

okay , display is working but another issue, i wrote command to display box from 0,0 to 256,64 and its only showing first 8 rows  u feel me, idk U8G2_SH1122_256X64_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/10, /* dc=*/9, /* reset=*/8);

void setup()
{
  u8g2.begin();
  u8g2.clearBuffer();
  u8g2.setDrawColor(1);
  //u8g2.setFont(u8g2_font_helvB08_tr);
  //u8g2.drawStr(0, 9, "Test");
  u8g2.drawBox(0,0,256,64);

  u8g2.sendBuffer();
}

olikraus, 2023-05-19:
You need to use the picture loop output because your did use the `_1_` constructor.

Use the firstPage / nextPage commands:

https://github.com/olikraus/u8g2/wiki/u8g2reference#carduino-example