olikraus/u8g2 #1991: Cannot get Contructor to work  properly for RA8835

Swejjj, 2022-10-24:
Hello,

I have been trying to get a 240x320 LCD with an RA8835 driver to work. I am using this constructor:

U8G2_RA8835_320X240_F_8080(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 9600 bytes]

When I run the code, I get the error that pinouts (d0-d7) are not declared in the scope. What I have already tried to do.

- Used the newest version of u8g2, tried a different constructor with the same driver: 

U8G2_RA8835_NHD_240X128_F_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7, /*enable=*/ 17, /*cs=*/ 14, /*dc=*/ 15, /*reset=*/ 16); // Connect /RD = E with +5V, enable is /WR = RW, FG with GND, 14=Uno Pin A0

Which does compile but does not run.

I am using an arduino mega 2560 to interface, I have also attached the code I am trying to run (Basic hello world). To whoever sees this, thank you for taking the time to look.

Here is the error log:

Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

HelloWorld:290:33: error: 'rotation' was not declared in this scope

 U8G2_RA8835_320X240_F_8080 u8g2(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 9600 bytes]

                                 ^~~~~~~~

C:\Users\AppData\Local\Temp\arduino_modified_sketch_323604\HelloWorld.ino:290:33: note: suggested alternative: 'static'

 U8G2_RA8835_320X240_F_8080 u8g2(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 9600 bytes]
[HelloWorldRA8835.txt](https://github.com/olikraus/u8g2/files/9848739/HelloWorldRA8835.txt)

                                 ^~~~~~~~

                                 static

HelloWorld:290:43: error: 'd0' was not declared in this scope

 U8G2_RA8835_320X240_F_8080 u8g2(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 9600 bytes]

                                           ^~

C:\Users\AppData\Local\Temp\arduino_modified_sketch_323604\HelloWorld.ino:290:43: note: suggested alternative: 'A0'

 U8G2_RA8835_320X240_F_8080 u8g2(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 9600 bytes]

                                           ^~

                                           A0

HelloWorld:290:47: error: 'd1' was not declared in this scope

 U8G2_RA8835_320X240_F_8080 u8g2(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 9600 bytes]

                                               ^~

C:\Users\AppData\Local\Temp\arduino_modified_sketch_323604\HelloWorld.ino:290:47: note: suggested alternative: 'A1'

 U8G2_RA8835_320X240_F_8080 u8g2(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 9600 bytes]

                                               ^~

                                               A1

HelloWorld:290:51: error: 'd2' was not declared in this scope

 U8G2_RA8835_320X240_F_8080 u8g2(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 9600 bytes]

                                                   ^~

C:\Users\AppData\Local\Temp\arduino_modified_sketch_323604\HelloWorld.ino:290:51: note: suggested alternative: 'A2'
[... 119 lines cut ...]
 U8G2_RA8835_320X240_F_8080 u8g2(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 9600 bytes]

                                                                                             ^~~~~

C:\Users\AppData\Local\Temp\arduino_modified_sketch_323604\HelloWorld.ino:290:93: note: suggested alternative: 'memset'

 U8G2_RA8835_320X240_F_8080 u8g2(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 9600 bytes]

                                                                                             ^~~~~

                                                                                             memset

HelloWorld:290:101: error: expected ',' or ';' before '[' token

 U8G2_RA8835_320X240_F_8080 u8g2(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 9600 bytes]

                                                                                                     ^

exit status 1

'rotation' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

olikraus, 2022-10-24:
What is the original source code here?

Swejjj, 2022-10-24:
> What is the original source code here?

Thanks for the quick reply! 

I am using the hello world example from the u8g2 library. u8g2>Fullbuffer>helloworld. Running it with no additional code.
I should note that it does not have the constructor listed for my LCD driver (RA8835 320x240, does have it for 240 x 128) but you do provide the constructor code for a 320 x 240 on your full list of constructor references which I added.

olikraus, 2022-10-24:
Ok, so how does your constructor line look like?

Swejjj, 2022-10-24:
> Ok, so how does your constructor line look like?

U8G2_RA8835_320X240_F_8080 u8g2(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 9600 bytes]

olikraus, 2022-10-24:
Look very wrong. Please check the existing commented constructors for proper arguments. See also the user manual. Please note that the square brackets describe optional arguments. You need to use or remove the arguments. Square brackets must be removed.

closed 2023-02-22 by olikraus