olikraus/u8g2 #1997: Platformio 2.33.15 arduino library has disappeared

jurgelenas, 2022-10-31:
Hello,

Our project depends on 2.33.15 and it has disappeared from the platformio registry. Can you please look into it?

olikraus, 2022-10-31:
The tag is here: https://github.com/olikraus/U8g2_Arduino/tree/2.33.15

I can not help with the platformio registry, but your ini file might include this:

`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

jurgelenas, 2022-10-31:
> The tag is here: https://github.com/olikraus/U8g2_Arduino/tree/2.33.15
>
> I can not help with the platformio registry, but your ini file might include this:
>
> `platformio.ini` [...]

But why 2.33.15 is gone from the platformio registry? Is this some kind of platformio issue and we need to contact them directly?

jurgelenas, 2022-10-31:
By the way we are using your library at https://www.expresslrs.org/3.0/

Really great project. Thanks for making it happen!

olikraus, 2022-11-01:
I am not related to platformio in any way, so I also don't know.

jurgelenas, 2022-11-01:
It was solved in https://github.com/ExpressLRS/ExpressLRS/pull/1924 on our side. Thanks for the help!

closed 2022-11-01 by jurgelenas