olikraus/u8g2 #1998: SH1108 128x160

fritolays, 2022-11-04:
Hello,

The SH1108 160x160 constructor does work with the screen I have, but things get cut off.
If you could kindly add the 128x160 resolution please and thanks.

[image]
[image]

olikraus, 2022-11-04:
Hmm... in the demo code from here: https://www.buydisplay.com/white-1-92-inch-128x160-oled-display-sh1108-controller-spi-i2c
they are using an offset of 16. Can you confirm this?

fritolays, 2022-11-04:
You are correct.
An offset of 15 no pixel visible.
Offset of 16, pixel visible on left side of screen.

[image]

olikraus, 2022-11-05:
I have created beta 2.34.5.
Can you test the "FlipMode" example? It should always show a solid border.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

fritolays, 2022-11-06:
The border looks to be correct.

[image]

olikraus, 2022-11-06:
The text changes 8 times, is the border always correct?

fritolays, 2022-11-06:
Sorry, yes the border is correct in all 8 iterations.

olikraus, 2022-11-06:
Super, then the new constructor should work for you. Thanks for testing.

closed 2022-11-19 by olikraus