olikraus/u8g2 #1999: Compiler whingeing in super-fussy mode

urbantigerau, 2022-11-05:
Hi Dev,
At the almost complete stage of many of my projects I like to do a review of the code to see what I missed, things like unused variables, failure to initialise variables, unused cases in a switch using enumerated types, and so on. These are useful to fix as they can cause unexpected side effects down the track. One of the ways I do this is to enable the compiler's most fussy mode regarding warnings. I am in the process of such a review right now, and some warnings from the compiler have surfaced from the u8g2 library.

See below, as copied from the IDE error window:
D:\New_Fred\Documents\Arduino\libraries\U8g2\src\clib\u8g2_polygon.c: In function 'pg_hline':

D:\New_Fred\Documents\Arduino\libraries\U8g2\src\clib\u8g2_polygon.c:219:10: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

   if ( y >= u8g2_GetDisplayHeight(u8g2) )  // does not work for 256x64 display???

          ^

D:\New_Fred\Documents\Arduino\libraries\U8g2\src\clib\u8g2_polygon.c:225:13: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

     if ( x1 >= u8g2_GetDisplayWidth(u8g2) )

             ^

D:\New_Fred\Documents\Arduino\libraries\U8g2\src\clib\u8g2_polygon.c:229:13: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

     if ( x2 >= u8g2_GetDisplayWidth(u8g2) )

             ^

D:\New_Fred\Documents\Arduino\libraries\U8g2\src\clib\u8g2_polygon.c:237:13: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

     if ( x2 >= u8g2_GetDisplayWidth(u8g2) )

             ^

D:\New_Fred\Documents\Arduino\libraries\U8g2\src\clib\u8g2_polygon.c:241:13: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

     if ( x1 >= u8g2_GetDisplayWidth(u8g2) )

             ^

All these seem to relate to comparing a signed value to an unsigned one.  I have not looked closely at the underlying source code, primarily because I might do more harm than good. My coding skills are OK, not expert.

Please note that this is not to seen as a complaint that requires instant rectification, as it may be that the compiler is just being excessively fussy, and the problem it refers to may not be practically ever seen. Please view it as information only. The library works very nicely. However, it may also be important, and I will leave that to your judgement. Thanks for a great library.

If you need further information about how I am using the library please ask.

Regards,

olikraus, 2022-11-05:
Thanks for reporting this. I will have a look.

olikraus, 2023-02-22:
same issue here: https://github.com/olikraus/u8g2/pull/2101

olikraus, 2023-06-06:
closed here, resolved via PR https://github.com/olikraus/u8g2/pull/2101

closed 2023-06-06 by olikraus