olikraus/u8g2 #2000: Cannot create fonts above 50px

jie326513988, 2022-11-05:
Hi
You can't create more than 50PX fonts with the font creation tool you provided. Is there a solution.
bdfconv and bdfconv_2_ 22 Not even.

rem bdfconv.exe -v -f 1 -m "48-58" CabinSketch-Regular_80.bdf -o helvb18_tf.c  -n u8g2_font_helvB18_tf -d CabinSketch-Regular_80.bdf
rem type helvb18_tf.c

bdfconv_2_22.exe -v -f 1 -m "48-58" CabinSketch-Regular_80.bdf -o helvb18_tf.c  -n u8g2_font_helvB18_tf -d CabinSketch-Regular_80.bdf
type helvb18_tf.c

olikraus, 2022-11-05:
Yes, there is a technical limitation in the format. If the compressed bitmap of the glyph exceeds 255 bytes, then you will get the above error. This can not be avoided without update of the font format. However an update of the font format is currently not intended.

jie326513988, 2022-11-07:
> 是的，格式存在技术限制。如果字形的压缩位图超过 255 字节，则会出现上述错误。如果不更新字体格式，这是无法避免的。但是，目前不打算更新字体格式。

OK, thank you for your answer. I'll think of another way.

closed 2023-02-22 by olikraus

tomtobback, 2023-11-15:
Hi, i've run into the same problem, using FontForge to create a 80px bdf, which still works, but does not really seem 80px high using it with u8g2. When i increase the bdf to 100px `bdfconv` fails as above. I can't really think of another way to get a large font of 100-120px, do you have any suggestion? The largest full font i find in u8g2 is 58px. I don't need it to be very sharp, is there any way to scale a smaller font into a larger one?