olikraus/u8g2 #2002: SSD1305-basad 128x32 display exhibiting two column left-shift

RickRae, 2022-11-06:
A [Waveshare Pico-OLED-2.23](https://www.waveshare.com/wiki/Pico-OLED-2.23) is behaving exactly like a setup suffering from SSD1306/SH1106 confusion: everything appears shifted two columns to the left with garbage in the rightmost two columns.  It is being driven in SPI mode by an Arduino Nano; test code is the basic u8g2 "Hello World!" demo sketch.

Manufacturer documentation claims this display uses an SSD1305 (132 pixels wide, same as an SH1106).  Both constructors U8G2_SSD1305_128X32_NONAME_1_4W_HW_SPI and _SW_ work (with the stated issue), while every SH1106 and SSD1306 constructor I've tried resulted in a blank display (excepting garbage in the rightmost column).  So it does appear it's either a SSD1305 or something (nearly) compatible with it.

Searches turned up lots of information on the SSD1306/SH1106 issue (including a post where the actual mechanism was described; thank you for that), but I've found almost nothing on the SSD1305.

Can you suggest an alternate constructor that might be more appropriate, or some diagnostics I can run to get a handle on the issue?  (I considered trying the manufacturer-provided demo code, but I doubt that would offer any hints regarding u8g2 configuration.)

Thanks!

olikraus, 2022-11-06:
Hm the sh1106 should work.

RickRae, 2022-11-06:
Here is the SH1106 I tried:
`U8G2_SH1106_128X64_NONAME_1_4W_HW_SPI u8g2(U8G2_R0, 10, 9); `

Compare versus the constructor that works (with a two column left shift):
`U8G2_SSD1305_128X32_NONAME_1_4W_HW_SPI u8g2(U8G2_R0, 10, 9);`

The software-emulated version "works" as well:
`U8G2_SSD1305_128X32_NONAME_1_4W_SW_SPI u8g2(U8G2_R0, 13, 11, 10, 9, 8);`

Happy to provide any information, or run whatever tests, that might help.

RickRae, 2022-11-20:
Any further thoughts on this?  I can afford to lose four columns, and the right-side garbage can be physically masked off.  Thus it's not a deal-breaker for this application.  But I may not have that flexibility in the next project, so it would be nice to get this resolved.

Thanks!

olikraus, 2022-11-20:
Oh, I guess I missed this somehow. 
I have a question: If you use the SSD1305 constructor, then there are 2 garbage columns on the left, correct?
Is this the same for SW and HW SPI?
What about the right side. Are two colums missing then?
It would be cool to have a picture, maybe from the u8g2 FlipMode example.

RickRae, 2022-11-20:
No worries.

`If you use the SSD1305 constructor, then there are 2 garbage columns on the left, correct?... What about the right side. Are two colums missing then?`

No, the output is shifted to the left, so there are two garbage columns on the _right_ (and data needs a horizontal offset to start at the left edge).  Similar to how I've seen the common SSD1306/SH1106 mix-up described.

`Is this the same for SW and HW SPI?`

Correct.  And I confirmed that by building FlipMode both ways just now.

`It would be cool to have a picture...`

Here is a screen from the actual application.  I intentionally cast a light on the front of the display, hoping that will help you to see the edges of the active area:

[image]

And per your request, FlipMode (both orientations):

[image]

[image]

Apologies for the potato shots, but good enough for this purpose, I suppose.

olikraus, 2022-11-20:
ok, great. Thanks for the pictures.
Please locate the following file and line in that file on your local hard disk:
https://github.com/olikraus/u8g2/blob/bed4e970523bf23329f7d3fdd4f0c42e44e94f47/csrc/u8x8_d_ssd1305.c#L152-L153

Change the default_x_offset value from 2 to 4: Will this help?
If flip mode is active, the probably flipmode_x_offset needs to be 0 or 4. Well I never know. Maybe you can test this.
Finally both values should be modified in such a way that the above flipmode test doesn't show any garbage.

I can also create a new constructor for your display based on the correct values.

RickRae, 2022-11-20:
I am out of time for today but wanted to at least let you know: The default_x_offset edit you suggested corrected the normal display.  I will work on the FlipMode aspect tomorrow and confirm the values for you (or at least report what I find, and we can go from there).

RickRae, 2022-11-21:
Here is the result of further testing. Changing default_x_offset to 4 centered the display for both normal and flipped orientations.  On the inverse video lines you can see the width of the white space before the "F" is the same in both orientations:

[image]

[image]

This is with leaving flipmode_x_offset untouched at 2.  Is that an expected outcome, or would you like me to run additional tests?

olikraus, 2022-11-21:
Can you also run the u8g2 (!) flip mode test? There should be a solid frame all the time. Thanks.

RickRae, 2022-11-21:
Hah, good call!  The left edge of the frame was missing at 180° and 270° rotations.  Final configuration which keeps the frame fixed in place and complete is: default_x_offset 4, flipmode_x_offset 0.

olikraus, 2022-11-22:
While trying to modify the code, i relized, that there is a constructor with the same configuration: 

```
U8G2_SSD1305_128X32_ADAFRUIT_1_4W_HW_SPI u8g2(U8G2_R0, 10, 9);
```

Can you check whether this constructor works for you?

RickRae, 2022-11-22:
Well now I just feel foolish.  I didn't try any of the "named" constructors because I assumed they were specific to the brands, and I didn't want to risk wrongly initializing something and possibly bricking the display.

I returned the parameters in u8x8_d_ssd1305.c to their original values and did a build to confirm the left-shift was back.  Then I swapped in the Adafruit constructor and did a follow-up build.

It worked perfectly.

Sorry to have wasted your time with this.  But perhaps this discussion will turn up in Web searches by others with the same issue and ultimately save you more time than I used?

Anyway: Thank you for both the support and for the excellent and comprehensive display libraries.

Rick

closed 2022-11-22 by RickRae

olikraus, 2022-11-22:
My mistake that I forgot those other constructors 🤪.

michaelosthege, 2023-06-24:
I also struggled with getting the Waveshare Pico OLED 2.23" to work and discovered this issue in the process.

First I tried to use SPI and got it to "work" with this configuration:
* `U8G2_SSD1305_128X32_ADAFRUIT_F_4W_HW_SPI display(U8G2_R0, PIN_SPI_SS, 8, 9);`
* Connecting Arduino Uno pin 8 to DC on the display board
* Connecting Arduino Uno pin 9 to RST on the display board

However, the **display got scrambled up again and again**, and this used 7 wires (VIN, GND, CLK, MOSI, CS, DC, RST) in total.

Therefore, I tried the **I2C interface with much more success:**

* `U8G2_SSD1305_128X32_NONAME_F_HW_I2C display(U8G2_R0, PIN_A4, PIN_A5);` gave the same two-column shift as described in the issue description
* `U8G2_SSD1305_128X32_ADAFRUIT_F_HW_I2C display(U8G2_R0, PIN_A4, PIN_A5);` worked perfectly 🎉

Next I'm trying to connect multiple of these Waveshare displays and will update my [example project for this](https://github.com/michaelosthege/display-i2c-example/blob/main/README.md).