olikraus/u8g2 #2005: Setting a new font

HajmamadAgha, 2022-11-09:
So I'm trying to make a new font, I used Fony for creating them and then used bdfconv to make it usable. but all I can see is a square and nothing else, I would appreciate it if you once for ever show us the guidline.

olikraus, 2022-11-09:
Which steps did you do in detail?
Which commandline options did you use?
Maybe also send you bdf file.

HajmamadAgha, 2022-11-10:
At first I designed my icon with fony, then typed the following command in cmd:
`bdfconv.exe -v -f 2 -th 2 -tv 2 -m "0-10" MyThing.bdf -o MyThing.c -n MyThing`
I got a c code, I used than directly in my sketch and the result is:
`const uint8_t BioPrinter[78] U8G2_FONT_SECTION("BioPrinter") = 
  "\1\0\3\4\5\5\1\4\6\21\20\0\371\0\0\0\0\0\0\0\0\0\61\0/\21\216\30\17'R\224"
  "B\202VU\304D\212Dd\42\32\31\42\13\12\11\12\303\26\24\22\24F\64\62D$\42S\311\24\325"
  "\245\220 \25)\5\0\0\0\4\377\377\0";
  `
but it doesn't show anything.

olikraus, 2022-11-10:
How did you print the gylphs in u8g2? I mean using encoding < 32 is little bit dangerouse, because only drawGylyph will work.
It might be better to use encodings with values higher or equal to 32 (=Space)

-f 2 is little bit unusual and also -th  and -tv may not work with -f 2
In general see here:
https://github.com/olikraus/u8g2/blob/d987608d40284f8efc75da202308442f953ad431/doc/faq.txt#L261-L276

closed 2023-02-22 by olikraus