olikraus/u8g2 #2006: Problem adding u8x8 Lib to my Atmel Studio 7

brunofmc50, 2022-11-10:
I am creating a project in C (yes it must be in C!) with Atmel Studio 7 that should read a temperature value by DHT22 and display on the 0.91" Oled 128x32. 
I was able to successfully add the u8g2 library where I can use functions such as: u8g2_DrawStr. However, as my goal is to show a temperature that is constantly changing I need to add something (that i don't know) to use the u8g2.printf function because the function u8g2_DrawStr just show a constant string. 

As I mentioned, I can visualize on my oled everything I write with the u8g2_DrawStr function. I added this **u8g2_Setup_ssd1306_i2c_128x32_univision_f(&u8g2, U8G2_R0, u8x8_byte_avr_hw_i2c, u8x8_avr_delay);**
and all worked perfect with u8g2 functions in C.
So what i need to add to my code to use u8g2.printf ? 
I tried to put this code line however i dont know the parameteres... 
void u8x8_Setup(u8x8_t *u8x8, u8x8_msg_cb display_cb, u8x8_msg_cb cad_cb, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);

Can you help me?

[image]

olikraus, 2022-11-10:
You need to convert a number to a string, which is not a u8g2 problem. Look out for sprintf or itoa.

brunofmc50, 2022-11-11:
Can you give me a simple example please? I'm not used to programming in C :/

closed 2023-02-22 by olikraus