olikraus/u8g2 #2007: Using KS0108 based GLCD with STM32F4 series

LALA-KHAN, 2022-11-11:
Hello
I recently designed a development baord that has a STM32F446Z on it as well as a KS0108 based GLCD.

I have tried a lot of ways to make the LCD work but I am stuck and this is one last try before I switch to another LCD.

I finally have been able to compile the code but it is not working at the moment.
Can anyone please guide me on how to port this library succesfully over to STM32
Thank you.

This is the u8g2_stm32f4.c file that I included in the main.

```#include "main.h"
#include "stm32f4xx_hal.h"
#include "u8g2.h"

#define DEVICE_ADDRESS 	0x3C
#define TX_TIMEOUT		100

extern I2C_HandleTypeDef hi2c3;

uint8_t u8x8_stm32_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	/* STM32 supports HW SPI, Remove unused cases like U8X8_MSG_DELAY_XXX & U8X8_MSG_GPIO_XXX */
	switch(msg)
	{
		case U8X8_MSG_GPIO_AND_DELAY_INIT:
		/* Insert codes for initialization */
		break;
		case U8X8_MSG_DELAY_MILLI:			/* ms Delay */
			HAL_Delay(arg_int);
		break;
		case U8X8_MSG_GPIO_D0:				// D0 or SPI clock pin: Output level in arg_int
			HAL_GPIO_WritePin(GPIOE, GPIO_PIN_9, arg_int);
	      break;
	    case U8X8_MSG_GPIO_D1:				// D1 or SPI data pin: Output level in arg_int
	    	HAL_GPIO_WritePin(GPIOE, GPIO_PIN_8, arg_int);
	      break;
	    case U8X8_MSG_GPIO_D2:				// D2 pin: Output level in arg_int
	    	HAL_GPIO_WritePin(GPIOE, GPIO_PIN_7, arg_int);
	      break;
	    case U8X8_MSG_GPIO_D3:				// D3 pin: Output level in arg_int
	    	HAL_GPIO_WritePin(GPIOG, GPIO_PIN_1, arg_int);
	      break;
	    case U8X8_MSG_GPIO_D4:				// D4 pin: Output level in arg_int
	    	HAL_GPIO_WritePin(GPIOG, GPIO_PIN_0, arg_int);
	      break;
	    case U8X8_MSG_GPIO_D5:				// D5 pin: Output level in arg_int
	    	HAL_GPIO_WritePin(GPIOF, GPIO_PIN_15, arg_int);
	      break;
	    case U8X8_MSG_GPIO_D6:				// D6 pin: Output level in arg_int
	    	HAL_GPIO_WritePin(GPIOF, GPIO_PIN_14, arg_int);
	      break;
	    case U8X8_MSG_GPIO_D7:				// D7 pin: Output level in arg_int
	    	HAL_GPIO_WritePin(GPIOF, GPIO_PIN_13, arg_int);
	      break;
	    case U8X8_MSG_GPIO_E:				// E/WR pin: Output level in arg_int
	    	HAL_GPIO_WritePin(GPIOE, GPIO_PIN_10, arg_int);
	      break;
	    case U8X8_MSG_GPIO_RESET:			// Reset pin: Output level in arg_int
	    	HAL_GPIO_WritePin(GPIOE, GPIO_PIN_11, arg_int);
	      break;
	    case U8X8_MSG_GPIO_CS1:				// CS1 (chip select) pin: Output level in arg_int
	    	HAL_GPIO_WritePin(GPIOF, GPIO_PIN_12, arg_int);
	      break;
	    case U8X8_MSG_GPIO_CS2:				// CS2 (chip select) pin: Output level in arg_int
	    	HAL_GPIO_WritePin(GPIOF, GPIO_PIN_11, arg_int);
	      break;
	    case U8X8_MSG_GPIO_I2C_CLOCK:		// arg_int=0: Output low at I2C clock pin
	    	HAL_GPIO_WritePin(GPIOA, GPIO_PIN_8, arg_int);
	      break;							// arg_int=1: Input dir with pullup high for I2C clock pin
	      HAL_GPIO_WritePin(GPIOC, GPIO_PIN_9, arg_int);
	    case U8X8_MSG_GPIO_I2C_DATA:			// arg_int=0: Output low at I2C data pin
	      break;							// arg_int=1: Input dir with pullup high for I2C data pin

	    default:
	      u8x8_SetGPIOResult(u8x8, 1);			// default return value
	      break;
	}
	return 1;
}

uint8_t u8x8_byte_ks0108(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  uint8_t i, b;
  uint8_t *data;

  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
      data = (uint8_t *)arg_ptr;
      while( arg_int > 0 )
      {
	b = *data;
	data++;
	arg_int--;
	for( i = U8X8_MSG_GPIO_D0; i <= U8X8_MSG_GPIO_D7; i++ )
	{
	  u8x8_gpio_call(u8x8, i, b&1);
	  b >>= 1;
	}

	u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->data_setup_time_ns);
	u8x8_gpio_call(u8x8, U8X8_MSG_GPIO_E, 1);
	u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->write_pulse_width_ns);
	u8x8_gpio_call(u8x8, U8X8_MSG_GPIO_E, 0);
      }
      break;

    case U8X8_MSG_BYTE_INIT:
      /* disable chipselect */
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
      /* ensure that the enable signal is low */
      u8x8_gpio_call(u8x8, U8X8_MSG_GPIO_E, 0);
      break;
    case U8X8_MSG_BYTE_SET_DC:
      u8x8_gpio_SetDC(u8x8, arg_int);
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      /* expects 3 bits in arg_int for the chip select lines */
      u8x8_byte_set_ks0108_cs(u8x8, arg_int);
      u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->post_chip_enable_wait_ns, NULL);
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->pre_chip_disable_wait_ns, NULL);
      u8x8_byte_set_ks0108_cs(u8x8, arg_int);
      break;
    default:
      return 0;
  }
  return 1;
}

uint8_t u8x8_byte_stm32_hw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	/* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
	static uint8_t buffer[32];
	static uint8_t buf_idx;
	uint8_t *data;

	switch(msg)
	{
	case U8X8_MSG_BYTE_SEND:
		data = (uint8_t *)arg_ptr;
		while( arg_int > 0 )
		{
			buffer[buf_idx++] = *data;
			data++;
			arg_int--;
		}
		break;
	case U8X8_MSG_BYTE_INIT:
		/* add your custom code to init i2c subsystem */
		break;
	case U8X8_MSG_BYTE_SET_DC:
		break;
	case U8X8_MSG_BYTE_START_TRANSFER:
		buf_idx = 0;
		break;
	case U8X8_MSG_BYTE_END_TRANSFER:
		if(HAL_I2C_Master_Transmit(&hi2c3, (DEVICE_ADDRESS << 1), buffer, buf_idx, TX_TIMEOUT) != HAL_OK) return 0;
		break;
	default:
		return 0;
	}
	return 1;
}
```

olikraus, 2022-11-11:
There are a lot of things to check and of course the amount of code is too less to say something.

 > I have tried a lot of ways to make the LCD work but I am stuck and this is one last try before I switch to another LCD.

In fact it might be a better idea to do so. Using a I2C or SPI 128x64 display is MUCH simpler than getting a KS0108 to work.

So let me list some potential hardware issues (from the previous > 2000 issues here):
- Did you correctly connect the var pot? 
- Is there a negative voltage converter available for your var pot?
- Did you use a proper voltage level converter for your digital lines?
- Did you wire everything correctly? Protoboards are always difficult, try to solder everything

As an initial test: You should be able to turn all pixel on, by moving your var pot from one position to the other.

LALA-KHAN, 2022-11-12:
I think I am going to keep the KS0108 on hold for now and just use the SH1106 that I have, unfortunately there is no ready to use example project of the port for u8g2 and my specific controller and Display. So I will just use the SSD1306 library for now until I get better at coding for STM32 and embedded sytems in gerneral. I will then try to port this library.
It would have been so much better to use the u8g2 library as I want to use the existing board.

olikraus, 2022-11-12:
👍

closed 2023-02-22 by olikraus