olikraus/u8g2 #2008: u8g2.getStrWidth() appears to return inconsistent values

ErikErikErikErikErik, 2022-11-13:
I get inconsistent values from passing the same const char* parameter to u8g2.getStrWidth() in two different locations. The parameter is a const private member in an object, thus it's never different, yet i get different results.

I've tried disabling U8G2_BALANCED_STR_WIDTH_CALCULATION, but it's the same outcome.

I realize this isn't much to go on, but wanted to see if someone has had the same problem.

Thanks in advance!

olikraus, 2022-11-13:
Did you change the font in between? The result depends on the current selected font.

https://github.com/olikraus/u8g2/wiki/u8g2reference#getstrwidth

ErikErikErikErikErik, 2022-11-13:
Well spotted! Thanks for your quick reply, that did the trick!

closed 2022-11-13 by ErikErikErikErikErik

olikraus, 2022-11-13:
👍