olikraus/u8g2 #2017: 如何在GPIO and Delay函数里设置菜单引脚？

SuJinkai, 2022-11-20:
我将u8g2移植到stm32上，但是我不知道如何设置菜单引脚，我无法控制菜单，u8x8_SetGPIOResult()不起作用。

SuJinkai, 2022-11-21:
I tried to write code like this, but only one pin can work. Why?

    case U8X8_MSG_GPIO_MENU_SELECT:
      //u8x8_SetGPIOResult(u8x8, /* get menu select pin state */ 1);
      break;
    case U8X8_MSG_GPIO_MENU_NEXT:
			if(HAL_GPIO_ReadPin(NEXT_GPIO_Port,NEXT_Pin)==GPIO_PIN_SET)
			{
				HAL_Delay(20);
				if(HAL_GPIO_ReadPin(NEXT_GPIO_Port,NEXT_Pin)==GPIO_PIN_SET)
				{
					u8x8_SetGPIOResult(u8x8, /* get menu next pin state */ 1);
					printf("ok!\n");
				}
			}
			else
			{
				u8x8_SetGPIOResult(u8x8, /* get menu next pin state */0);
			}
      //u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ 1);
      break;
    case U8X8_MSG_GPIO_MENU_PREV:
      //u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ 1);
      break;
    case U8X8_MSG_GPIO_MENU_HOME:
      //u8x8_SetGPIOResult(u8x8, /* get menu home pin state */ 1);
      break;
    default:
      //u8x8_SetGPIOResult(u8x8, 1);			// default return value
      break;

olikraus, 2022-11-21:
you probably need to repeat the code also for the other messages.

SuJinkai, 2022-11-21:
I'm really glad you can get back to me because I'm desperate for an answer, I've tried doing this but all the menu pins don't work, is it my code that's wrong?

    case U8X8_MSG_GPIO_MENU_SELECT:
      u8x8_SetGPIOResult(u8x8, /* get menu select pin state */ HAL_GPIO_ReadPin(SELECT_GPIO_Port,SELECT_Pin));
      break;
    case U8X8_MSG_GPIO_MENU_NEXT:
      u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ HAL_GPIO_ReadPin(NEXT_GPIO_Port,NEXT_Pin));
      break;
    case U8X8_MSG_GPIO_MENU_PREV:
			u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ HAL_GPIO_ReadPin(PREV_GPIO_Port,PREV_Pin));
      break;
    case U8X8_MSG_GPIO_MENU_HOME:
      u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ HAL_GPIO_ReadPin(HOME_GPIO_Port,HOME_Pin));
      break;
    default:
      u8x8_SetGPIOResult(u8x8, 1);			// default return value
      break;

SuJinkai, 2022-11-21:
> 您可能还需要为其他消息重复代码。

I'm really glad you can get back to me because I'm desperate for an answer, I've tried doing this but all the menu pins don't work, is it my code that's wrong?

    case U8X8_MSG_GPIO_MENU_SELECT:
      u8x8_SetGPIOResult(u8x8, /* get menu select pin state */ HAL_GPIO_ReadPin(SELECT_GPIO_Port,SELECT_Pin));
      break;
    case U8X8_MSG_GPIO_MENU_NEXT:
      u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ HAL_GPIO_ReadPin(NEXT_GPIO_Port,NEXT_Pin));
      break;
    case U8X8_MSG_GPIO_MENU_PREV:
			u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ HAL_GPIO_ReadPin(PREV_GPIO_Port,PREV_Pin));
      break;
    case U8X8_MSG_GPIO_MENU_HOME:
      u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ HAL_GPIO_ReadPin(HOME_GPIO_Port,HOME_Pin));
      break;
    default:
      u8x8_SetGPIOResult(u8x8, 1);			// default return value
      break;

olikraus, 2022-11-21:
It looks correct to me

olikraus, 2022-11-21:
20MB rar archive? Seriously not. you can post small code fragments here, but for anything else, I don't have time.

SuJinkai, 2022-11-22:
> 20MB rar archive? Seriously not. you can post small code fragments here, but for anything else, I don't have time.

i 'm sorry. This is the complete gpio_ and_ Delay, what's wrong with the code?I don't know why. I can't adjust the layout, so it's a little ugly. I've tried my best to adjust the format.

uint8_t u8x8_stm32_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  switch(msg)
  {
    case U8X8_MSG_GPIO_AND_DELAY_INIT:	// called once during init phase of u8g2/u8x8
      break;							// can be used to setup pins
    case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second
      break;    
    case U8X8_MSG_DELAY_100NANO:		// delay arg_int * 100 nano seconds
			__NOP();
      break;
    case U8X8_MSG_DELAY_10MICRO:		// delay arg_int * 10 micro seconds
			for (uint16_t n = 0; n < 320; n++)
			{
				__NOP();
			}
      break;
    case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli second
			HAL_Delay(arg_int);
      break;
    case U8X8_MSG_DELAY_I2C:				// arg_int is the I2C speed in 100KHz, e.g. 4 = 400 KHz
      break;							// arg_int=1: delay by 5us, arg_int = 4: delay by 1.25us
    case U8X8_MSG_GPIO_D0:				// D0 or SPI clock pin: Output level in arg_int
    //case U8X8_MSG_GPIO_SPI_CLOCK:
      break;
    case U8X8_MSG_GPIO_D1:				// D1 or SPI data pin: Output level in arg_int
    //case U8X8_MSG_GPIO_SPI_DATA:
      break;
    case U8X8_MSG_GPIO_D2:				// D2 pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_D3:				// D3 pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_D4:				// D4 pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_D5:				// D5 pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_D6:				// D6 pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_D7:				// D7 pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_E:				// E/WR pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_CS:				// CS (chip select) pin: Output level in arg_int
			//if (arg_int) HAL_GPIO_WritePin(CS_GPIO_Port, CS_Pin, GPIO_PIN_RESET);
			//else HAL_GPIO_WritePin(CS_GPIO_Port, CS_Pin, GPIO_PIN_SET);
		
			if (arg_int) HAL_GPIO_WritePin(CS_GPIO_Port, CS_Pin, GPIO_PIN_SET);
			else HAL_GPIO_WritePin(CS_GPIO_Port, CS_Pin, GPIO_PIN_RESET);
      break;
    case U8X8_MSG_GPIO_DC:				// DC (data/cmd, A0, register select) pin: Output level in arg_int
			if (arg_int) HAL_GPIO_WritePin(DC_GPIO_Port, DC_Pin, GPIO_PIN_SET);
			else HAL_GPIO_WritePin(DC_GPIO_Port, DC_Pin, GPIO_PIN_RESET);
      break;
    case U8X8_MSG_GPIO_RESET:			// Reset pin: Output level in arg_int
			if (arg_int) HAL_GPIO_WritePin(RST_GPIO_Port, RST_Pin, GPIO_PIN_SET);
			else HAL_GPIO_WritePin(RST_GPIO_Port, RST_Pin, GPIO_PIN_RESET);
      break;
    case U8X8_MSG_GPIO_CS1:				// CS1 (chip select) pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_CS2:				// CS2 (chip select) pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_I2C_CLOCK:		// arg_int=0: Output low at I2C clock pin
      break;							// arg_int=1: Input dir with pullup high for I2C clock pin
    case U8X8_MSG_GPIO_I2C_DATA:			// arg_int=0: Output low at I2C data pin
      break;							// arg_int=1: Input dir with pullup high for I2C data pin
    case U8X8_MSG_GPIO_MENU_SELECT:
      u8x8_SetGPIOResult(u8x8, /* get menu select pin state */ HAL_GPIO_ReadPin(SELECT_GPIO_Port,SELECT_Pin));
      break;
    case U8X8_MSG_GPIO_MENU_NEXT:
      u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ HAL_GPIO_ReadPin(NEXT_GPIO_Port,NEXT_Pin));
      break;
    case U8X8_MSG_GPIO_MENU_PREV:
			u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ HAL_GPIO_ReadPin(PREV_GPIO_Port,PREV_Pin));
      break;
    case U8X8_MSG_GPIO_MENU_HOME:
      u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ HAL_GPIO_ReadPin(HOME_GPIO_Port,HOME_Pin));
      break;
    default:
      u8x8_SetGPIOResult(u8x8, 1);			// default return value
      break;
  }
  return 1;
}

olikraus, 2022-11-22:
Thanks for providing a more readable form. However: It all looks ok to me. Did you debug, whether you HAL functions work as expected?

SuJinkai, 2022-11-23:
> Thanks for providing a more readable form. However: It all looks ok to me. Did you debug, whether you HAL functions work as expected?

After debugging, I found the problem because the pin is low by default. Now it has been solved. Thank you very much! If no one helps me, I may not be able to find out the problem.

closed 2022-11-23 by SuJinkai

Amirmahdisoltani, 2024-03-17:
> After debugging, I found the problem because the pin is low by default. Now it has been solved. Thank you very much! If no one helps me, I may not be able to find out the problem.

Hello, I've been struggling for a few days to set up the pins for the menu. If you've solved this issue, could you share the code you wrote with me?
I set up the example from the link below with Arduino on the blue pill board, as shown in the picture. 
link: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/MUIDemo/MUIDemo.ino
[image]
However, when I try to compile the same code in CubeIDE, I encounter the following errors:"
-----------
10:47:40 **** Incremental Build of configuration Debug for project uc1701-hw-l431cct6 ****
make -j6 all 
arm-none-eabi-gcc "../Core/Src/main.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32L431xx -c -I../Core/Inc -I../Drivers/STM32L4xx_HAL_Driver/Inc -I../Drivers/STM32L4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32L4xx/Include -I../Drivers/CMSIS/Include -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Core/Src/main.d" -MT"Core/Src/main.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/main.o"
arm-none-eabi-gcc -o "uc1701-hw-l431cct6.elf" @"objects.list"   -mcpu=cortex-m4 -T"C:\Users\Desktop\  workspace\stm32\uc1701-hw-l431cct6\STM32L431CCTX_FLASH.ld" --specs=nosys.specs -Wl,-Map="uc1701-hw-l431cct6.map" -Wl,--gc-sections -static --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -Wl,--start-group -lc -lm -Wl,--end-group
C:/ST/STM32CubeIDE_1.13.2/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.win32_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: ./Core/Src/main.o:(.rodata.muif_list+0x28): undefined reference to `u8g2_font_streamline_food_drink_t'
C:/ST/STM32CubeIDE_1.13.2/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.win32_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: ./Core/Src/u8g2/mui_u8g2.o: in function `mui_u8g2_draw_button_utf':
C:/Users/Desktop/workspace/stm32/uc1701-hw-l431cct6/Debug/../Core/Src/u8g2/mui_u8g2.c:234: undefined reference to `u8g2_DrawButtonUTF8'
C:/ST/STM32CubeIDE_1.13.2/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.win32_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: ./Core/Src/u8g2/mui_u8g2.o: in function `mui_u8g2_u8_chkbox_wm_pi':
C:/Users/Desktop/workspace/stm32/uc1701-hw-l431cct6/Debug/../Core/Src/u8g2/mui_u8g2.c:1599: undefined reference to `u8g2_DrawButtonFrame'
C:/ST/STM32CubeIDE_1.13.2/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.win32_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: ./Core/Src/u8g2/mui_u8g2.o: in function `mui_u8g2_u8_radio_wm_pi':
C:/Users/Desktop/workspace/stm32/uc1701-hw-l431cct6/Debug/../Core/Src/u8g2/mui_u8g2.c:1656: undefined reference to `u8g2_DrawButtonFrame'
C:/ST/STM32CubeIDE_1.13.2/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.win32_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: ./Core/Src/u8g2/mui_u8g2.o: in function `mui_u8g2_u8_opt_radio_child_w1_pi':
C:/Users/Desktop/workspace/stm32/uc1701-hw-l431cct6/Debug/../Core/Src/u8g2/mui_u8g2.c:1844: undefined reference to `u8g2_DrawButtonFrame'
C:/ST/STM32CubeIDE_1.13.2/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.win32_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: ./Core/Src/u8g2/mui_u8g2.o: in function `mui_u8g2_u8_opt_child_wm_pi':
C:/Users/Desktop/workspace/stm32/uc1701-hw-l431cct6/Debug/../Core/Src/u8g2/mui_u8g2.c:1882: undefined reference to `u8g2_DrawButtonUTF8'
C:/ST/STM32CubeIDE_1.13.2/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.win32_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: ./Core/Src/u8g2/mui_u8g2.o: in function `mui_u8g2_u16_list_child_w1_pi':
C:/Users/Desktop/workspace/stm32/uc1701-hw-l431cct6/Debug/../Core/Src/u8g2/mui_u8g2.c:2203: undefined reference to `u8g2_DrawButtonFrame'
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:65: uc1701-hw-l431cct6.elf] Error 1
"make -j6 all" terminated with exit code 2. Build might be incomplete.

10:47:42 Build Failed. 8 errors, 0 warnings. (took 1s.802ms)

I think I am receiving these errors for the following two lines in the main function of the program:
-
mui_Init(&mui, (void *)u8g2_GetU8x8(&u8g2), fds_data, muif_list, sizeof(muif_list)/sizeof(muif_t));
	mui_GotoForm(&mui,/* form_id= */ 0, /* initial_cursor_position= */ 0);

I would appreciate it if you could guide me. Thank you.

olikraus, 2024-03-31:
Looks like you have missed to link all the c code with your project.