olikraus/u8g2 #2020: Cant add ssd1306 128x64

brunofmc50, 2022-11-21:
hi, i cant add my oled display 128x64, what should i do?
[image]

olikraus, 2022-11-21:
you need to remove the line 
``` C++
u8g2_t u8g2;
```

brunofmc50, 2022-11-21:
It keeps giving error... My oled display is 128x64.
[image]

olikraus, 2022-11-21:
1. You should call the constructor outside the main function (like in your initial post)
2. You need to include <U8g2lib.h>

But it looks like you are not in an Arduino environment, so you need to follow the instructions from here: https://github.com/olikraus/u8g2/wiki/u8g2setupc

Also have a look at my example here: https://github.com/olikraus/u8g2/blob/master/sys/avr/as7/main.c

Which was referenced here: https://github.com/olikraus/u8g2/wiki/u8g2as7

closed 2023-02-22 by olikraus