olikraus/u8g2 #2030: Support for CH1116

TheCatfather92, 2022-11-29:
This is in continuation of issue #1808, where someone had trouble initializing CH1116 based display. A project of mine involves this driver-based display, and I could not find much proper info. After a lot of digging, it seems to be closely related to the SSD1309. Here is the link to the article discussing compatibility. There is a full list of commands as well as an initialization sequence. Perhaps that would be useful to people in the coming days.

https://www.winstar.com.tw/technology/oled/45.html

[image]

[image]

olikraus, 2022-11-29:
Thanks for the information. After reading the referenced web page, here are some more comments from my end:

- CH1116 only supports page addressing mode, which is ok, because u8g2 will always try to use page addressing mode.
- The IREF topic might be an issue, because this is not considered in SSD1309 constructor
- The CH1116 includes a charge pump (internal VPP), which is the biggest issue: If a display vendor chooses to use the internal charge pump, then the CH1116 display will not work with the SSD1309 constructor (because the required code to activate the charge pump is missing in the SSD1309).

Charge Pump Usage:

- CH1116 displays will work with the SSD1309 constructor if the display hardware uses external VPP (IREF is still an issue)
- CH1116 displays will not work with the SSD1309 constructor if the display hardware uses the internal charge pump for VPP generation. In this case we would need a new constructor, which enables the charge pump (or the u8g2 user needs to send an additional code to activate the charge pump).

I found a CH1116 display, which will not work with the existing SD1309 constructor:
https://datasheet.lcsc.com/lcsc/2109031030_Shenzhen-Allvision-Tech-N130-2864KLBLG01-H30_C2890598.pdf

For this display, the charge pump needs to be enabled. So we could use the SSD1309 constructor with the following additional command (of course not tested):

```
  u8g2.sendF("cac", 0x0ad, 0x08b, 0x032);
```

I personally would wait with the creation of a specific CH1116 constructor until someone comes along with such a display. We probably also need two versions of that constructor: With and without internal charge pump.

olikraus, 2022-11-29:
Looking a little bit further, it looks like that the CH1116 is kind of compatible with the SH1106. Indeed the charge pump commands are identical between SH1106 and SSD1309. This leads to the following conclusion:

- CH1116 with external VPP power: Use SSD1309 constructor
- CH1116 with internal VPP generation (Charge Pump): Use SH1106 constructor

olikraus, 2022-11-29:
I found a CH1116 display: https://de.aliexpress.com/item/1005004911560811.html

TheCatfather92, 2022-11-29:
Thank you for the very fast and concise response. My project uses a display very similar to the first one you found (white variant).
https://www.lcsc.com/product-detail/OLED-Displays-Modules_Shenzhen-Allvision-Tech-N130-2864KSWLG01-H30_C2890597.html
It is connected to Rp2040 (and an ESP32 optionally) using the I2C bus and is configured to use the internal charge pump. I'll test the recommendations here and update. Unfortunately, the parts are still arriving from china (expected by mid december).
Until then I hope others find the info here useful.

closed 2023-02-22 by olikraus