olikraus/u8g2 #2038: Can't get st7567 I2C lcd to work

EliasVP, 2022-12-04:
Hello,

I bought an ST7567 I2C lcd for a project that I'm working on (https://nl.aliexpress.com/item/1005004237282931.html?spm=a2g0o.order_list.order_list_main.4.f8d779d2mq1xUS&gatewayAdapt=glo2nld), I have tested it with this library (https://github.com/luetee/ST7567S_128X64_I2C) and the graphicstest worked right away. I however would love to use your library but I just can't seem to get it to diplay anything. I have tried pretty much every st7567 constructor in the library and tested with several arduino's (nano, uno, leonardo, nodeMCU), and I don't know what I am doing wrong (might just be somthing stupid that I missed). In the reviews of the lcd on aliexpress there are several people who claim that they used "U8G2_ST7567_JLX12864_F_SW_I2C u8g2(U8G2_R0, /* klok = */ 5, /* data = */ 4, /* reset = */ U8X8_PIN_NONE);" as constructor but it doesn't seem to work for me.

thanks in advance

closed 2022-12-04 by EliasVP

olikraus, 2022-12-04:
So finally u8g2 did work?

EliasVP, 2022-12-04:
Yes, 5 min after I send my message I found the problem. So now it works like a charm.