olikraus/u8g2 #2039: Problems with SSD1322 on ESP32

Cynicroute, 2022-12-04:
Hi everyone,

Bear with me - i'm a novice hobbyist and just started working with displays for a vehicle project.

I have the SSD1322 configured to 4SPI and I've had it running well, first on a Nano clone, then on an ESP8266. But I've decided to switch to ESP32 to read OBDii over BT.

As I mentioned, I know the display works. I used u8g2 on both the Nano and 8266 with no issues. Also, I know the current pin assignments on the ESP32 are ok because I got the display working in the micropython environment. But when I switch to the more-familiar-to-me Arduino environment, the display does nothing. 

What am I missing? TIA for any suggestions!

```
#include <Arduino.h>
#include <SPI.h>
#include <Wire.h>
#include <U8g2lib.h>

/*
 * SSD1322      ESP32
 * 1 GND         GND
 * 2 VIN           3.3V
 * 4 CLK           5
 * 5 DATA        16
 * 14 DC          22
 * 15 RES         23
 * 16 CS          19
 */

//-- SSD1322 OLED Constructor (16-BIT ENABLED)

//U8G2_SSD1322_NHD_256X64_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 5, /* data=*/ 16, /* cs=*/ 19, /* dc=*/ 22, /* reset=*/ 23);
U8G2_SSD1322_NHD_256X64_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 19, /* dc=*/ 22, /* reset=*/ 23);

void setup() {
  u8g2.begin();
  u8g2.enableUTF8Print();  
  Serial.begin(115200);
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.drawHLine(108, 34, 164);
    u8g2.setFont(u8g2_font_finderskeepers_tf);
    u8g2.setCursor(0, 38);
    u8g2.print("SCREEN TEST");

    Serial.print("Upload worked...");
  } while ( u8g2.nextPage() );
  delay(1000);
}

```

olikraus, 2022-12-04:
Will this help?

``` C++
U8G2_SSD1322_NHD_256X64_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 19, /* dc=*/ 22, /* reset=*/ 23);

void setup() {
  u8x8_SetPin(u8g2.getU8x8(), U8X8_PIN_SPI_CLOCK, 5);
  u8x8_SetPin(u8g2.getU8x8(), U8X8_PIN_SPI_DATA, 16);
  u8g2.begin();
  u8g2.enableUTF8Print();  
  Serial.begin(115200);
}
```

Cynicroute, 2022-12-04:
Thanks Oli - but no luck :(

I did come across an alternate pin assignment that works in SW-SPI however. No luck with that configuration in HW-SPI though.

```
U8G2_SSD1322_NHD_256X64_1_4W_SW_SPI u8g2(U8G2_R2, /* clock=*/ 19, /* data=*/ 23, /* cs=*/ 22, /* dc=*/ 16, /* reset=*/ 17);
```

olikraus, 2022-12-04:
There is a default assignment of the ESP32 environment to some specific pins. Probably you need to use this default assignment  for the SPI pins.

Cynicroute, 2022-12-05:
I ended up running an SPI scan and got these results.

MOSI: 23
MISO: 19
SCK: 18
SS: 5

SPI Scan:
```
/*
  Rui Santos
  Complete project details at https://RandomNerdTutorials.com/esp32-spi-communication-arduino/
  
  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files.
  
  The above copyright notice and this permission notice shall be included in all
  copies or substantial portions of the Software.
*/

//Find the default SPI pins for your board
//Make sure you have the right board selected in Tools > Boards
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.print("MOSI: ");
  Serial.println(MOSI);
  Serial.print("MISO: ");
  Serial.println(MISO);
  Serial.print("SCK: ");
  Serial.println(SCK);
  Serial.print("SS: ");
  Serial.println(SS);  
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

Based on that, I changed my pin config to this contstructor:
```//-- SSD1322 OLED Constructor (16-BIT ENABLED)

U8G2_SSD1322_NHD_256X64_1_4W_SW_SPI u8g2(U8G2_R2, /* clock=*/ 18, /* data=*/ 23, /* cs=*/ 5, /* dc=*/ 19, /* reset=*/ 17);
//U8G2_SSD1322_NHD_256X64_1_4W_HW_SPI u8g2(U8G2_R2, /* cs=*/ 5, /* dc=*/ 19, /* reset=*/ 17);
```

I presume these are the default HW-SPI assignments for this board.

If anyone else is having a similar issue with these rubbish NodeMCU ESP32 boards, try the [10uf capacitor trick](https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/) to get your sketch uploaded.

For pin assignments, these have worked for me with SW-SPI:

``` 
SSD1322             ESP32
1 GND               GND
2 VIN               3.3V
4 CLK               5 / 19 / 18
5 DATA/MOSI         16 / 23 / 23
14 DC               22 / 16 / 19
15 RES              23 / 17 / 23
16 CS               19 / 22 / 5
 ```
Your board may differ, so run the SPI scan and see what comes back.

closed 2023-02-22 by olikraus