olikraus/u8g2 #2041: Dynamic Memory Satured

talakasse, 2022-12-07:
Hello everyone,

I recently started to use u8g2 to make two SSD1306 work on an single ATMEGA328P.
My code is pretty heavy but u8g2lib is filling my dynamic memory to 96%.
I only have 80 bytes left for my others vars.

Here's the message : 
The sketch uses 11658 bytes (37%) of program storage space. The maximum is 30720 bytes.
**Global variables use 1936 bytes (94%) of dynamic memory, leaving 112 bytes for local variables. The maximum is 2048 bytes.**
Low available memory, stability issues may occur.

I tried to use all the commands in the documentation but the minimum i reached is 94%, i also tried to delete some useless commands and fonts in the lib files but it didn't work.

If anybody have an idea it would be helpful.

tlk

olikraus, 2022-12-07:
Which constructor did you use? You should use the page buffer mode!

talakasse, 2022-12-07:
i use : 
U8G2_SSD1306_128X64_NONAME_F_HW_I2C display1(U8G2_R0, 5,4,U8X8_PIN_NONE);
U8G2_SSD1306_128X64_NONAME_F_HW_I2C display2(U8G2_R0, 5,4,U8X8_PIN_NONE);

what is buffer mode ?

olikraus, 2022-12-07:
> what is [page] buffer mode ?

The solution to your problem:

https://github.com/olikraus/u8g2/blob/2104af8263c53d44a33ad8a24b54143d39256980/doc/faq.txt#L134-L142

In Arduino IDE you will also find a lot of example for the page buffer mode.

Manual: https://github.com/olikraus/u8g2/wiki/u8g2reference#firstpage

For more details, see here: https://github.com/olikraus/u8glib/wiki/tpictureloop

talakasse, 2022-12-07:
ok thanks ! :)

talakasse, 2022-12-12:
#include <Arduino.h>
#include <Wire.h>
#include <U8g2lib.h>
#include <virtuabotixRTC.h>

U8G2_SSD1306_128X64_NONAME_1_HW_I2C display1(U8G2_R0, 5,4,U8X8_PIN_NONE);
virtuabotixRTC myRTC(5,6,7);

#define BUTTON1 2
#define BUTTON2 4
#define PTENTIO_MENU A7
#define MAXx 128
#define MINx 5
#define MAXy 10
#define MINy 50

unsigned long CTIME = 0, LTIME = 0;
int bouton_buz = 8;
int j ; 
float moyenne = 0 ;
float tab [50] ;
float maximum;
float nbbpm ; 
long Time0;
long Time; 
int cycle=1;
int x1=0; 
int y1=40;
int x2=0; 
int y2=40;
float lastnbbpm; 
int seq; 
int echelle; 
int valpot;
const int pot = A2 ;
int bpm;
const int capteurcardiaque = A0 ;
const int ledv = 10;
const int ledr = 9;
const int ledj = 11;
int tps=0;
const int buzzer=12;
int tpssec;
int position[100];
int u = 0;
int OLD_POS = 0;
unsigned short int curve[MAXx - MINx] = {0};
int x = 0 ;
// Variable de langue
bool langue = 0;

void setup (){
  display1.setI2CAddress(0x78);
  display1.begin();
  display1.enableUTF8Print();
  display1.setFont(u8g2_font_6x10_tr);
  display1.clearDisplay();
}

void loop (){
   if(x1>127){
    
    x1=1;
    x2=x1;
  }
  moyenne=0;
  for(j=0;j<50;j++){//on rempli le tableau de 50 valeurs lues par le capteur cardiaque 
  tab[j] = analogRead(capteurcardiaque) ; 
  moyenne = moyenne + tab[j] ;
  delay(1);
  }
  moyenne = (float)(moyenne / 50) ;//On calcul la moyenne de ces 50 valeurs
  //on cherche le seuil
  for (j=0;j<50;j++){
    tab[j]=(tab[j]-moyenne);//On soustrait à ces valeurs la moyenne
  }
 maximum = 0;
 for(j=0;j<50;j++){//Dans cette boucle on determine le maximum du tableau
  if (tab[j]> maximum){
    maximum = tab[j];
  }
 }
  for(j=0;j<50;j++){//On divise le valeurs par le maximum si elle est superieur a 0 pour avoir des valeurs entre 1 et 0
    if ((tab[j]/maximum)>=0){
      tab[j] =(float)tab[j]/(float)maximum;
    }
    else{//Si le resultat de la division est négatif on donne à la valeur 0
    tab[j]=0.00;
    }
  }
  for(j=0;j<50;j++){//ON compte combien de valeurs sont au dessus du seuil pour determiner le BPM
    if(tab[j]>0.95){
        nbbpm=nbbpm+1;
    }
   }
  /////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
  valpot=analogRead(pot); //on relie le potentiometre et la valeur du BPM
  echelle=map(valpot,0,1023,5,20);//on convertit les valeurs de 0 a 1023 en valeurs de 5 a 20 pour l'echelle 

  Time0 = ((float) millis()/(float) 1000);
  Time=5*cycle;
  if((Time0/(Time))>=1){
    nbbpm=(nbbpm*12)/25;
    lastnbbpm=nbbpm;
    cycle++;
    seq= 128/((nbbpm/60));
    Serial.println(nbbpm);
  }

display1.firstPage();
do {
  
    display1.setFontMode(1);
  display1.setDrawColor(1);

  //int value = 0, max = 0, min = 0;
  if(x1%seq==0)
    {//Si le reste de la division de x1 par seq est egale à 0, 
    y2=40-(nbbpm/6);
    y1=40;
    display1.drawLine(x1, y1,x2,y2);//On dessine une ligne entre le point (x1,y1) et (x2,y2)
    x1=x2;//x1 prend la valeur de x2
    x2++;//x2 egal x2+1
    
    }
  
  if (x1%seq==1)
    {//Si le reste de la division de x1 par seq est egale à 1, 
    y1=40-(nbbpm/6);
    y2=50;
    display1.drawLine(x1, y1,x2,y2);//On dessine une ligne entre le point (x1,y1) et (x2,y2)
    x1=x2;//x1 prend la valeur de x2
    x2++;//x2 egal x2+1
    
    }

  if (x1%seq==2)
    {//Si le reste de la division de x1 par seq est egale à 2, 
    y1= 50;
    y2=40;
    display1.drawLine(x1, y1,x2, y2);//On dessine une ligne entre le point (x1,y1) et (x2,y2)
    x1=x2;//x1 prend la valeur de x2
    x2++;//x2 egal x2+1
    
    }

  if((x1%seq!=0) && (x1%seq!=1) && (x1%seq!=2))
    {//On entre dans ce block si les conditions avant ne sont pas vérifiées
    y1=40;
    y2=40;
    display1.drawLine(x1, y1,x2, y2);//On dessine une ligne entre le point (x1,y1) et (x2,y2)
    y1=y2;
    x1=x2;//x1 prend la valeur de x2
    x2++;//x2 egal x2+1
    }
  
  
  
  //COURBE MAUVAISE
  //COURBE(&value, &max, &min);
  /*Serial.println(value);
  value = map(value, min, max, MAXy, MINy);
  if(x > /*MAXx - MINx20){x = 0;Serial.println("RST : " + String(x));};
  /*curve[x] = value;
  Serial.println("INC : " + String(x));
  Serial.println("VXX : " + String(x + MINx));
  Serial.println("VAL : " + String(curve[x]));
  */
    //display1.firstPage();  
  
  //AFFICHAGE DE L'HEURE ET DU BPM 
  
    display1.setFontMode(1);
    display1.setDrawColor(2);
    display1.setCursor(0, 7);
    display1.print(lastnbbpm);
    display1.setCursor(35, 7);
    display1.print("BPM");
    display1.setCursor(76, 7);
    display1.println(myRTC.hours);
    display1.setCursor(90,7);
    display1.println(":");
    display1.setCursor(95,7);
    display1.println(myRTC.minutes);
    display1.setCursor(109,7);
    display1.println(":");
    display1.setCursor(114,7);
    display1.println(myRTC.seconds);
    // Ligne Horizontale et fleche 
    display1.drawLine(0, 55 - 5, 128, 55 - 5);
    display1.drawLine(128 - 4, 55 - 5 - 3, 128 - 1, 55 - 5);
    display1.drawLine(128 - 4, 55 - 5 + 3, 128 - 1, 55 - 5);
    // ligne verticale et fleche 
    display1.drawLine(0 + 5, 55, 0 + 5, 10);
    display1.drawLine(0 + 5, 10, 0 + 5 - 3, 10 + 4);
    display1.drawLine(0 + 5, 10, 0 + 5 + 3, 10 + 4);
    
    
    
    
    
   

  /*display1.drawPixel(128 - 1, 31);
  display1.drawPixel(x, lastnbbpm);
  display1.drawRFrame(0,9,128,64-9,004);
  display1.drawRFrame(0,9,128,55,004);*/
      
  }
  while(display1.nextPage());
  x1++;

  
}

talakasse, 2022-12-12:
I try several ways and I really can't

olikraus, 2022-12-12:
My suggestion is:
 * Store the data in an array
 * Draw the array content

Unfortunately your code is unreadable, still I do not see how you draw the content of "tab".

talakasse, 2022-12-13:
ok I see where the problem is, thanks again for your help