olikraus/u8g2 #2045: How to use not the  defaults for SDA and SCL for I2C

barnycloud, 2022-12-10:
I must use other pins for SDA and SCL (SDA= D3 / pin 0 and SCL D4/ pin 2) using I2C in my esp8266. The (no-name) OLED-display has a SH1006 controller. The following constructors do not work,, i.e. the display remains dark:
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, /* clock=/ SCL, / data=*/ SDA);
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=/ U8X8_PIN_NONE, / clock=/ SCL, / data=*/ SDA);
SDA and SCL are defined as follows in the Arduino-IDE (used as editor) code:
#define SDA 0 // Pin 0 = D3
#define SCL 2 // // Pin 2 = D4
What went wrong? Any help appreciated.

olikraus, 2022-12-10:
You could start with the SW I2C constructor

barnycloud, 2022-12-10:
Thank You for the quick response. Bur I do not find a SW constructor for SH1106 NONAM in the list. UsingWire.begin(SDA,SCL) it is possible to use non-default pins for the I2C. But I would prefer to define SDA and SCL pins in the u8g2 constructor. No chance?

olikraus, 2022-12-10:
Which list did you check? The official list is here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp

It would probably make sense to start with this constructor:
```
U8G2_SH1106_128X64_NONAME_F_SW_I2C(rotation, clock, data);
U8G2_SH1106_128X64_NONAME_F_SW_I2C(rotation, clock, data, reset);
```
or the low RAM variant:
```
U8G2_SH1106_128X64_NONAME_1_SW_I2C(rotation, clock, data);
U8G2_SH1106_128X64_NONAME_1_SW_I2C(rotation, clock, data, reset);
```

barnycloud, 2022-12-10:
I have tried this code:
U8G2_SH1106_128X64_NONAME_F_SW_I2C  u8g2(U8G2_R0, SCL, SDA, /* reset=*/);
Here an error code appeared in the Arduino IDE: expected primary-expression before ')' token

U8G2_SH1106_128X64_NONAME_F_SW_I2C  u8g2(U8G2_R0,/* reset=*/  SCL, SDA);
Diaplay remains dark.

barnycloud, 2022-12-10:
This code works:
U8G2_SH1106_128X64_NONAME_F_SW_I2C  u8g2(U8G2_R0, 2, 0, U8X8_PIN_NONE);
Sorry, but I did not realize that /* */ means commented out.
Thanks for your effort!

closed 2023-02-22 by olikraus