olikraus/u8g2 #2046: u8g2_t Struct Possible Indeterminate Initialisation

chrissbarr, 2022-12-11:
Hello,

I've encountered some unexpected behaviour with this library. I have it resolved in my application, but wanted to share the details as it might be worth following up.

## Problem

Some relevant details of my setup:
Platform: ESP32-S3 with PSRAM
U8G2 instance constructed dynamically / at runtime (I think this is relevant), using internal SRAM or external PSRAM (trying both).

I am drawing a simple screen with text only. I am constructing the U8G2 instance like so:

```
display = std::make_unique<U8G2_SSD1309_128X64_NONAME0_F_4W_HW_SPI>(
  U8G2_R0, 
  pin::DISPLAY_CS, 
  pin::DISPLAY_DC, 
  pin::DISPLAY_RESET);
```

Depending on when in my code I construct U8G2, the vertical alignment of text drawn via `drawStr` changes. 

I've tracked this down to the struct member `u8g2->font_height_mode`. My reading is there are three valid values:

```
#define U8G2_FONT_HEIGHT_MODE_TEXT 0
#define U8G2_FONT_HEIGHT_MODE_XTEXT 1
#define U8G2_FONT_HEIGHT_MODE_ALL 2
```

I do not have a call in my code to any of the `setFontRefHeightXYZ` functions, so was just using whichever the default value is.

Most of the time, the behaviour I am seeing is as if `font_height_mode` is defaulting to `U8G2_FONT_HEIGHT_MODE_ALL`. 

If I change the order in which I am constructing other items in my code (before / after U8G2 constructor as above), the default behaviour of `u8g2->font_height_mode` changes to `U8G2_FONT_HEIGHT_MODE_TEXT` or `U8G2_FONT_HEIGHT_MODE_XTEXT`. I had come to rely on `U8G2_FONT_HEIGHT_MODE_ALL` as the default, and only after checking the documentation (https://github.com/olikraus/u8g2/wiki/u8g2reference#setfontrefheighttext) did I realise the default I should be expecting is `U8G2_FONT_HEIGHT_MODE_TEXT`, which further confused me.

Because the behaviour varied as I jostled memory around, I suspected there was a reference to uninitialised memory taking place.

Looking through the code, I see that the `u8g2_t u8g2` member of `class U8G2` is defined as so:

```
class U8G2
...
  protected:
    u8g2_t u8g2;
...
```

My reading is that this means when the U8G2 class is constructed, the `u8g2_t u8g2` struct will not be initialised, and therefore the members of that struct may be filled with whatever data was in the memory the struct now occupies. 

If the memory is previously unused (say, when U8G2 is compile-time constructed), those values are likely to be zero, but if the memory has been used (as in the case where some dynamic allocation / deallocation has taken place before constructing the U8G2 instance) then the values may be anything. 

Notably the code that uses `font_height_mode` checks something like `if (font_height_mode == U8G2_FONT_HEIGHT_MODE_TEXT) {} elseif (font_height_mode == U8G2_FONT_HEIGHT_MODE_XTEXT) {} else {}`. This explains why I am most often seeing behaviour matching `U8G2_FONT_HEIGHT_MODE_ALL`, as that's the fall-through case. So any random data in `font_height_mode` is most likely going to match U8G2_FONT_HEIGHT_MODE_ALL, unless it just so happens to be 0 or 1.

## Resolution
For my usage, it's sufficient so far to call `setFontRefHeightAll();` after constructing `U8G2`. I haven't reviewed the rest of the `u8g2_struct` to see what possible effects uninitialised memory could have, but I don't see any other visible problems.

I think it would be a good idea to resolve this somehow so the behaviour is deterministic / so that all required memory is initialised correctly. I'm unsure of how to do this in C and C++ - if it was only C++, you could probably do:

```
struct u8g2_struct
{
...
  uint8_t font_height_mode{};
or
  uint8_t font_height_mode = 0;
...
}
``` 
Etc., probably repeated for most struct members, along with default-constructing the U8G2 class member:
```
class U8G2
...
  protected:
    u8g2_t u8g2{};
...
```

But I am unsure if this would work with all versions of C and C++ supported by the library.

I wanted to write this down because I got stuck for a few days chasing it through. Hoping that even if closed at least it could be a helpful reference if someone else has a similar experience.

olikraus, 2022-12-11:
Thanks for the detailed description and nice finding. I also assume, that this is a bug. 

What I figured out so far is this:
`font_height_mode` is not set to 0 inside 
https://github.com/olikraus/u8g2/blob/bdc8d2beda5dc472a6e4d38ada6de33060615273/csrc/u8g2_setup.c#L71

So I assume this is the bug here: We need to assign 0 to `font_height_mode` inside `u8g2_SetupBuffe()`.

olikraus, 2022-12-11:
I have created a beta release. Can you test, whether this works.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

chrissbarr, 2022-12-11:
Ah, I completely missed that setup function. Yes, looks like initialising `font_height_mode` there should take care of the problem. Agree this is a good solution.

Thanks for the quick reply and quick fix! I am testing the beta release via PlatformIO now and will update shortly...

chrissbarr, 2022-12-11:
Yes, I can confirm that you have fixed it! I have run the following test:

```
// fill heap with array of known non-zero value
constexpr int rsize = 3000;
uint8_t* rarray = new uint8_t[rsize];
for (int i = 0; i < rsize; i++) {
    rarray[i] = 0xFF;
}
// delete array / free heap
delete [] array;
// construct U8G2
display = std::make_unique<U8G2_SSD1309_128X64_NONAME0_F_4W_HW_SPI>(U8G2_R0, pin::DISPLAY_CS, pin::DISPLAY_DC, pin::DISPLAY_RESET);
// Check value of font_height_mode 
Serial.printf("display->getU8g2()->font_height_mode = %d\n", display->getU8g2()->font_height_mode);
```

Using PlatformIO latest version I get:
```
display->getU8g2()->font_height_mode = 255
```

Using your zip / Github master I get:
```
display->getU8g2()->font_height_mode = 0
```

So looks great!

olikraus, 2022-12-11:
Thanks again for the bug report and the testing on the solution. 
I will close this ticket.

closed 2022-12-11 by olikraus