olikraus/u8g2 #2050: Support for third SPI interface on Teensy boards

bazmonk, 2022-12-14:
I'm interested in using U8g2 with SSD1306 and SH1106 128x64 displays on the third SPI that the Teensy4.1 offers.  The main HW SPI interface works well and I see there's a constructor to call the second interface already.

It looks like all that might be required is copying and tweaking `u8x8_byte_arduino_2nd_hw_spi` in https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp to use SPI2 instead of SPI1 and such, and adding the proper constructors.

My questions are:

* Am I underestimating what would be required to add this feature?  Is there a "gotcha" I'm not seeing yet?
* Assuming I did this, any special rules to submitting a pull request to your repo?  Would you want this upstream?

Thanks!

-Basil

olikraus, 2022-12-14:
Your suggested tweaking should work as expected. Putting this into the u8g2 repo is more difficult because of the introduction of a 3rd_hw_spi and also the automatic constructor generation (https://github.com/olikraus/u8g2/blob/master/tools/codebuild/codebuild.c) Additionally the code needs to check whether the 3rd interface is available. All in all it is some effort and I don't have time for this at the moment :-(

bazmonk, 2022-12-14:
Thanks for the reply.

I’m waiting on a part and then will probably try to make a go of this in January.  If I can, I’m willing to put in the effort necessary to make it acceptable upstream/acceptable into the main repository.

bazmonk, 2022-12-26:
Just FYI, I did get the 3rd SPI interface to work.  It ended up being just a matter of creating a `U8G2_SH1106_128X64_NONAME_F_3RD_4W_HW_SPI` exactly like `U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI`, in U8g2lib.h, and a 'u8x8_byte_arduino_3rd_hw_spi' exactly like `u8x8_byte_arduino_hw_spi` (not 2nd_hw_spi, oddly enough... not sure if that works on Teensy but the first does so I mirrored it), except SPI2 instead of SPI in U8x8lib.cpp/h.

Since it is literally just that SPI->SPI2 change, I think I can understand incorporating it into codebuild.c. Where is the code the checks if the interface is available that I'd have to handle for it to be acceptable upstream?

olikraus, 2022-12-26:
> Where is the code the checks if the interface is available that I'd have to handle for it to be acceptable upstream?

Here:

https://github.com/olikraus/u8g2/blob/f22a5a84209645f65c6ece0661c7f7518e7c8e9e/cppsrc/U8x8lib.h#L150-L154

We probably need another section like this:

``` C++
#ifdef SPI_INTERFACES_COUNT
#if SPI_INTERFACES_COUNT > 2
#define U8X8_HAVE_3RD_HW_SPI
#endif
#endif
```

Happy Holiday!

closed 2023-02-22 by olikraus

bazmonk, 2023-02-23:
Fair enough :-). Since getting it to work at all this way I've been too busy using it to circle back to making it upstream-acceptable.  Perhaps later when dev calms down.  Thx for your help with it.

olikraus, 2023-02-23:
👍