olikraus/u8g2 #2051: SSD1362 256x64 driver Winstar WEA025664A

olikraus, 2022-12-14:
### Discussed in https://github.com/olikraus/u8g2/discussions/2043

<div type='discussions-op-text'>

<sup>Originally posted by **provitek** December  9, 2022</sup>
Hi,
I wanted to make project with WInstar WEA025664A 256x64 oled display. SSD1362 controller.
unfortunately it is not directly compatible with u8g2.
Does anyone know what controller it is compatible with or closest to?
What would need to be changed?
Could it be added in the next release?
https://www.winstar.com.tw/pl/products/oled-module/graphic-oled-display/wea025664a.html
THX</div>

olikraus, 2022-12-14:
SSD1362 doesn't need the 0x5c to enable RAM write:

https://github.com/olikraus/u8g2/blob/cd5b4704a873a375882917c97cd33f4470b379cb/csrc/u8x8_d_ssd1322.c#L205 

The logic level for the DC line for command arguments is inverted. As a result we need "cad_001" for SSD1362 (instead of "cad_011").

https://github.com/olikraus/u8g2/blob/cd5b4704a873a375882917c97cd33f4470b379cb/tools/codebuild/codebuild.c#L1847

olikraus, 2022-12-14:
Proposed init sequence:
``` C++
static const uint8_t u8x8_d_ssd1322_256x64_init_seq[] = {
    
  U8X8_DLY(1),
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_DLY(1),
  
  
  U8X8_C(0xae),		                /* display off */
  U8X8_C(0XFD), //Set Command Lock
  U8X8_C(0X12), //(12H=Unlock,16H=Lock)

  U8X8_C(0XAE), //Display OFF(Sleep Mode)

  U8X8_C(0X15), //Set column Address
  U8X8_C(0X00), //Start column Address
  U8X8_C(0X7F), //End column Address

  U8X8_C(0X75), //Set Row Address
  U8X8_C(0X00), //Start Row Address
  U8X8_C(0X3F), //End Row Address

  U8X8_C(0X81), //Set contrast
  U8X8_C(0x2f),

  U8X8_C(0XA0), //Set Remap
  U8X8_C(0Xc3),

  U8X8_C(0XA1), //Set Display Start Line
  U8X8_C(0X00),

  U8X8_C(0XA2), //Set Display Offset
  U8X8_C(0X00),

  U8X8_C(0XA4), //Normal Display

  U8X8_C(0XA8), //Set Multiplex Ratio
  U8X8_C(0X3F),

  U8X8_C(0XAB), //Set VDD regulator
  U8X8_C(0X01), //Regulator Enable

  U8X8_C(0XAD), //External /Internal IREF Selection
  U8X8_C(0X8E),

  U8X8_C(0XB1), //Set Phase Length
  U8X8_C(0X22),

  U8X8_C(0XB3), //Display clock Divider
  U8X8_C(0XA0),

  U8X8_C(0XB6), //Set Second precharge Period
  U8X8_C(0X04),

  U8X8_C(0XB9), //Set Linear LUT

  U8X8_C(0XBc), //Set pre-charge voltage level
  U8X8_C(0X10), //0.5*Vcc

  U8X8_C(0XBD), //Pre-charge voltage capacitor Selection
  U8X8_C(0X01),

  U8X8_C(0XBE), //Set cOM deselect voltage level
  U8X8_C(0X07), //0.82*Vcc

  U8X8_C(0XAF), //Display ON

  U8X8_DLY(1),					/* delay 2ms */

  
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};
```

olikraus, 2022-12-14:
@provitek I have created a new beta 2.34.8 with a new constructor `U8G2_SSD1362_WS_256X64_1_4W_HW_SPI`

Can you test whether this works. If not: Please send a screenshot of the display content.
You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

provitek, 2022-12-15:
thanks,
unfortunately I can't do it for arduino right now. I copied csrc to my project in STM32cubeIDE, modified main.c, set 16 bit mode.
```c
u8g2_Setup_ssd1362_ws_256x64_f(&u8g2, U8G2_R0, u8x8_byte_stm32_hw_spi, u8x8_stm32_gpio_and_delay);
````
[image]

provitek, 2022-12-15:
I don't knows, maybe it helps:
functions from winstar demo:
```c
//-------------------------------------------------
//          Write Data Function
//-------------------------------------------------
void Write_data(uint8_t datap)
{
	OLED_CS_LOW;
	OLED_DC_HIGH;

	HAL_SPI_Transmit(&hspi2, &datap, 1, 100);

	OLED_DC_LOW;
	OLED_CS_HIGH; //	LCM_CS1 = 1;
}
//-------------------------------------------------
//     WRITE_COMMAND
//-------------------------------------------------
void Write_command(uint8_t command)
{

	OLED_CS_LOW;
	OLED_DC_LOW;

	HAL_SPI_Transmit(&hspi2, &command, 1, 100);

	OLED_DC_HIGH;
	OLED_CS_HIGH; //	LCM_CS1 = 1;
}
```c

olikraus, 2022-12-15:
great, looks better already.
I have updated https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1362.c
Can you test the new code again and send me a picture of it?

provitek, 2022-12-15:
much better now :)
[image]

olikraus, 2022-12-15:
well, yes, but I think the remap is still somehow wrong. I need more help from your end. Can you locate this line in your code:

https://github.com/olikraus/u8g2/blob/ad26a8f68c8d1e5b250085f8b25d41d7fb64842a/csrc/u8x8_d_ssd1362.c#L438-L440

Then replace 0xc3 by one of the following values: 0x03 0x83 0x13 0x93

Will any value improve the picture? Or is there even a value which makes it perfect?

It looks like the remap is another huge difference to the SSD1322, so we need to kind of reverse engineer the correct values.

provitek, 2022-12-15:
I'll check tonight and let you know

provitek, 2022-12-15:
Hi, 
I have tested with values as you give: (first - bitmap displayed via demo Winstar code, second - u8g2) 
0xC3
[image]
[image]

0x03
[image]
[image]

0x13
[image]
[image]

0x83
![0x83-bitmap](https://user-images.githubusercontent.com/50798040/207984529-f014488b-622b-
[image]
4710-8e9a-4dce34c221f8.jpg)

0x93
[image]
[image]

provitek, 2022-12-15:
Winstar demo code:
```c
//=========================================ȡģ��ʾ����
void White_IMAGE(unsigned char *p)
{
	unsigned int x,y;

	Write_command(0X15); //Set column Address
	Write_command(0X00); //Start column Address
	Write_command(0X7F); //End column Address

	Write_command(0X75); //Set Row Address
	Write_command(0X00); //Start Row Address
	Write_command(0X3F); //End Row Address
	
	for(y = 0;y < 64;y ++)
	{   
    	for(x = 0;x < 32;x ++)
		{
			write_4_data(*p++);
		}		
	} 
}
void write_4_data(unsigned char dispdata)		 //?4?����
{ 
	unsigned char i,temp0,temp1,temp2[4]; 

	temp0=dispdata;
	for(i=0;i<4;i++)
		{
   	    temp1=0xc0&temp0;		 // temp0�c0xc0���c��temp1�ĽY������߃�λ���ֲ�?�����Nλ��0�F
		switch(temp1)			 //�l���x��
			{
			case 0x00:			 //temp1=0x00
			temp2[i]=0x00;
			break;
			case 0x40:			 //temp1=0x40
			temp2[i]=0x0f;
			break;
    		case 0x80:			 //temp1=0x80
			temp2[i]=0xf0;		 
			break;
			case 0xc0:			 //temp1=0xc0
			temp2[i]=0xff;
			break;
			default:                           
   			break;
 			}
		temp0=temp0<<2;			 //���ƃ�λ
		Write_data(temp2[i]);				 //����?��һ�M������4?��
		}
}

//=========================================ȫ��/ȫ����ʾ����
void White_IMAGE1(unsigned char dat)
{
	unsigned int x,y;

	Write_command(0X15); //Set column Address
	Write_command(0X00); //Start column Address
	Write_command(0X7F); //End column Address

	Write_command(0X75); //Set Row Address
	Write_command(0X00); //Start Row Address
	Write_command(0X3F); //End Row Address
	
	for(y = 0;y < 64;y ++)
	{   
    	for(x = 0;x < 128;x ++)
		{
			Write_data(dat);
		}		
	} 
}

//=========================================	50%������ʾ����
void White_IMAGE2(unsigned char dat1,unsigned char dat2)
{
	unsigned int x,y,z;

	Write_command(0X15); //Set column Address
	Write_command(0X00); //Start column Address
	Write_command(0X7F); //End column Address

	Write_command(0X75); //Set Row Address
	Write_command(0X00); //Start Row Address
	Write_command(0X3F); //End Row Address
	
	for(y = 0;y < 32;y ++)
	{
		for(z=0;z<2;z++)
		{
    		for(x = 0;x < 32;x ++)
			{
				Write_data(dat1);
			}
			for(x = 0;x < 32;x ++)
			{
				Write_data(dat2);
			}		
		}  
	}
	for(y = 0;y < 32;y ++)
	{
		for(z=0;z<2;z++)
		{
    		for(x = 0;x < 32;x ++)
			{
				Write_data(~dat1);
			}
			for(x = 0;x < 32;x ++)
			{
				Write_data(~dat2);
			}		
		}  
	}
}

olikraus, 2022-12-16:
Thanks for your effort. I think I need to order one of the displays. Looks like more effort to understand the internal memory architecture.
Is this one of the items? https://de.aliexpress.com/item/32259159336.htmlhttps://de.aliexpress.com/item/32259159336.html

olikraus, 2022-12-16:
I have ordered the OLED. It will arrive in 4 weeks.

provitek, 2022-12-16:
Thank you! My pleasure :)
Sorry I couldn't help.
Its different display, hope it will work. Fingers crossed :)
mine is: https://www.winstar.com.tw/pl/products/oled-module/graphic-oled-display/wea025664a.html
THX
[image]

olikraus, 2022-12-16:
I think there is something I missed.i need to debug the display controller more closely.

olikraus, 2023-01-02:
Added support for my ordered display: [image]

All updates should be in the u8g2 repo now.

olikraus, 2023-01-02:
I have also created a new beta release 2.34.11...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

olikraus, 2023-01-02:
Note: I have change the constructor name a little bit: U8G2_SSD1362_256X64_1_4W_HW_SPI

https://github.com/olikraus/u8g2/wiki/gallery#3-jan-2023-ssd1362-256x64-and-206x36

closed 2023-01-02 by olikraus

provitek, 2023-01-03:
It works! :)
Thank you! 
[image]

olikraus, 2023-01-03:
Nice :-)

olikraus, 2023-12-08:
A new issue might be better. Also a picture, constructor, code, Arduino Board...

NixieGuy, 2023-12-11:
Sorry, all this is new for me. I'll start again. (Opening an issue makes me feel bad, like I'm bothering _everyone_ for no reason) Also, it's related to code in this thread.

So, while trying to use the screen MDOB256064D1Y-YS that also uses the SSD1306, using an STM32F103C8T6, Platform.io on Visual Studio Code.  I could not make it work, until I notticed that the descriptor written in this thread was:

U8G2_SSD1362_256X64_1_4W_HW_SPI

But the SSD1362 examples in the library had a 1 substituted by an F.

U8G2_SSD1362_256X64_F_4W_HW_SPI

 When I changed that in my code the screwn worked beautifully.

[image]

IFYBB, 2025-04-08:
Hello everyone. 

I'm trying to display text on the display WEA025664AWAP3N (SSD1362). I use the recommendations from the message https://github.com/olikraus/u8g2/issues/2051#issuecomment-1369275575, But that doesn't solve the problem.
Code:
```
#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>
#include <Wire.h>

U8G2_SSD1362_256X64_1_4W_HW_SPI u8g2(U8G2_R0, 10, 9, 8); // [page buffer, size = 256 bytes] for Arduino Mega

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(5, 5,"Hello!");
  } while ( u8g2.nextPage() );
}
```

Nothing appears on the display. Where am I going wrong?