olikraus/u8g2 #2053: Bug in u8g2_DrawLine() when distance is longer than 255 pixels (when U8G2_16BIT is needed)

spachner, 2022-12-16:
Hi Oli,

there is a bug in u8g2_DrawLine() when distance is longer than 256 pixels when U8G2_16BIT is needed for larger displays. Results in wrong slope of line. 

u8g2_DrawLine(x, 0, 0, 255, 255) works as intended.
u8g2_DrawLine(x, 0, 0, 256, 256) is wrongly rendered.

```
for( x = x1; x <= x2; x++ )
  {
    if ( swapxy == 0 ) 
      u8g2_DrawPixel(u8g2, x, y); 
    else 
      u8g2_DrawPixel(u8g2, y, x); 
    err -= (uint8_t)dy;
    if ( err < 0 ) 
    {
      y += (u8g2_uint_t)ystep;
      err += (u8g2_uint_t)dx;
    }
  }

```

```
for( x = x1; x <= x2; x++ )
  {
    if ( swapxy == 0 ) 
      u8g2_DrawPixel(u8g2, x, y); 
    else 
      u8g2_DrawPixel(u8g2, y, x); 
    err -= dy; // fix
    if ( err < 0 ) 
    {
      y += (u8g2_uint_t)ystep; 
      err += (u8g2_uint_t)dx;
    }
  }

```

regards

spachner

olikraus, 2022-12-16:
Nice finding. I will check. Thanks!

olikraus, 2022-12-17:
I think it should be fixed now. Thanks for the bug description.

closed 2022-12-17 by olikraus