olikraus/u8g2 #2057: bdfconv for Linux server

stonez56, 2022-12-21:
Hi,
I know there is a Windows bdfconv.exe in the \tools\ directory. 
However, I'm using Linux.  Is there a bdfconv Linux version that can be ran from the Linux command line?

Thank you,
Stonez56

stonez56, 2022-12-23:
In the Linux, I just went into the directory /tools/font/bdfconv/
and then run $>make command
Then I got the executable **bdfconv** binary under Linux.

Problem solved!

Stonez56

olikraus, 2022-12-23:
Excuse, I missed to answer this, but great you figured out by yourself 😊

stonez56, 2022-12-23:
@olikraus Thanks for this great u8g2 contribution to the community!
I'm glad that I figured it out myself!

BTW, I've made how to use u8g2 library with SSD1306 0.9" OLED in a YouTube channel, [one in Chinese](https://youtu.be/Ksdwd87JpPs), the [other in English](https://youtu.be/-bntyE8UYTY). 
Is it okay to place YouTube link here? If not, I will remove these URLs.
Hope these are helpful to u8g2 users!

Stonez56

olikraus, 2022-12-23:
> Is it okay to place YouTube link here? 

Of course. Great video , I already commented it as "ucglib".
Maybe you also want to create a "show and tell" entry in the discussions board, which refers to your video.

stonez56, 2022-12-24:
Great! I will create an entry in the "Show and tell" discussion board!

stonez56, 2022-12-27:
@olikraus I have another question, under the directory /tools/font/bdf  there are so many .bdf files
Is there a list where I can tell which .bdf belongs to what language and what font? 
For example, I would like to use Japanese, Chinese, Korean, and Hindi on the OLED. Which .bdf should I use with bdfconv to convert?

For example, I used this command:
./tools/font/bdfconv/bdfconv -v ./tools/font/bdf/unifont.bdf -b 0 -f 1 -M ./myChineseFont.map -d  ./tools/font/bdf/7x13.bdf -n u8g2_font_unifont_myfonts -o u8g2_font_unifont_myfonts.h  

I can generate quite a few Chinese, Japanese, and few Korean, but no Hindi. 

Thanks!
Stonez56

olikraus, 2022-12-27:
> Is there a list where I can tell which .bdf belongs to what language and what font?

There is no such list. A bdf file is just a list of glyphs. Each glyph is described by its unicode number. Which unicode numbers are included in the bdf file is not visible from outside and depends fully on the author. Of course you can open the bdf file in a text editor and check the content manually.

"bdfconv" is not only a conversion tool, but also a unicode selection tool. You can provide unicode ranges or individual unicode numbers which should be converted. Of course only existing unicode glyphs will be converted.

> I can generate quite a few Chinese, Japanese, and few Korean, but no Hindi.

For Hindi you probably need devanagari glyphs. The the first step is to get the uncode numbers for those glyphs (see for example here: https://en.wikipedia.org/wiki/Devanagari). Then you need to apply those glyphs to bdfconv via -m or -M command. Finally you can check the conversion output, whether the intended glyphs are converted or not.
If they are not converted, then the intended glyphs are not part of the bdf file. In case of unifont.bdf, devanagari glyphs should be included.

closed 2023-02-22 by olikraus