olikraus/u8g2 #2058: STM32F070 U8G2

vTaskCreate, 2022-12-22:
Hi;
I have JLX16080G-370-PC160 * 80 high dot matrix COG and it's have ST75256. I'm trying to use u8g2 but I couldn't, it's only seen blank screen, I tried sw_i2c, 4w spi sw and hw. 
Here is my code, would you tell me what's wrong? 
Thanks 
```
	uint8_t u8x8_stm32_gpio_and_delay(U8X8_UNUSED u8x8_t *u8x8,
		U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int,
		U8X8_UNUSED void *arg_ptr)
	{
	  switch (msg)
	  {
	  case U8X8_MSG_GPIO_AND_DELAY_INIT:
		HAL_Delay(1);
		break;
	  case U8X8_MSG_DELAY_MILLI:
		HAL_Delay(arg_int);
		break;
	  case U8X8_MSG_GPIO_DC:
		HAL_GPIO_WritePin(CD_LCD_GPIO_Port, CD_LCD_Pin, arg_int);
		break;
	  case U8X8_MSG_GPIO_RESET:
		HAL_GPIO_WritePin(LCD_RESET_GPIO_Port, LCD_RESET_Pin, arg_int);
		break;
	  }
	  return 1;
	}
	uint8_t u8x8_byte_4wire_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int,
		void *arg_ptr)
	{
	  switch (msg)
	  {
	  case U8X8_MSG_BYTE_SEND:
		HAL_SPI_Transmit(&hspi2, (uint8_t *) arg_ptr, arg_int, 10000);
		break;
	  case U8X8_MSG_BYTE_INIT:
		break;
	  case U8X8_MSG_BYTE_SET_DC:
		HAL_GPIO_WritePin(CD_LCD_GPIO_Port, CD_LCD_Pin, arg_int);
		break;
	  case U8X8_MSG_BYTE_START_TRANSFER:
		  HAL_GPIO_WritePin(LCD_CS_GPIO_Port, LCD_CS_Pin, GPIO_PIN_RESET);
		break;
	  case U8X8_MSG_BYTE_END_TRANSFER:
		  HAL_GPIO_WritePin(LCD_CS_GPIO_Port, LCD_CS_Pin, GPIO_PIN_SET);
		break;
	  default:
		return 0;
	  }
	  return 1;
	}
	/* USER CODE END 0 */
	
	/**
	  * @brief  The application entry point.
	  * @retval int
	  */
	int main(void)
	{
	  /* USER CODE BEGIN 1 */
		u8g2_t u8g2;
	  /* USER CODE END 1 */
	
	  /* MCU Configuration--------------------------------------------------------*/
	
	  /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
	  HAL_Init();
	
	  /* USER CODE BEGIN Init */
	
	  /* USER CODE END Init */
	
	  /* Configure the system clock */
	  SystemClock_Config();
	
	  /* USER CODE BEGIN SysInit */
	  SystemCoreClockUpdate();
	  /* USER CODE END SysInit */
	
	  /* Initialize all configured peripherals */
	  MX_GPIO_Init();
	  MX_SPI2_Init();
	  /* USER CODE BEGIN 2 */
	  HAL_GPIO_WritePin(LCD_CS_GPIO_Port, LCD_CS_Pin, GPIO_PIN_SET);
	  HAL_GPIO_WritePin(BACKLIGHT_GPIO_Port, BACKLIGHT_Pin, GPIO_PIN_RESET);
	  u8g2_Setup_st75256_jlx19296_2(u8g2, rotation, byte_cb, gpio_and_delay_cb)(&u8g2, U8G2_R0, u8x8_byte_4wire_hw_spi,u8x8_stm32_gpio_and_delay);
	  //u8g2_Setup_st75256_i2c_jlx19296_1(&u8g2, U8G2_R0, u8x8_byte_sw_i2c,u8g2_gpio_and_delay_stm32);
	  u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,
	  u8g2_SetPowerSave(&u8g2, 0); // wake up display
	
	
	  HAL_Delay(1000);
	
	  /* USER CODE END 2 */
	
	  /* Infinite loop */
	  /* USER CODE BEGIN WHILE */
	  while (1)
	  {
		/* USER CODE END WHILE */
	
		/* USER CODE BEGIN 3 */
			u8g2_FirstPage(&u8g2);
			do
			{
			  u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
			  u8g2_DrawStr(&u8g2, 0, 15, "Hello World!");
			  u8g2_DrawCircle(&u8g2, 64, 40, 10, U8G2_DRAW_ALL);
			} while (u8g2_NextPage(&u8g2));
	  }
	  /* USER CODE END 3 */
	}
	```

olikraus, 2022-12-22:
Maybe start with stm32duino to check whether your hardware works.

vTaskCreate, 2022-12-24:
Hi,

Thanks for the advice. Finally I put something on the LCD so I know the hardware is ok, as I mentioned before I'm using JLX16070G-370 which is not listed on the constuructor list. 
I used U8G2_ST75256_JLX19296_1_4W_SW_SPI but the resolution's not correct for me. 
[image]

is there any way change resolution ? 

btw thanks for the library. I'll put all project to github.

cheers.

olikraus, 2022-12-25:
> is there any way change resolution ?

Of course, but it is not that simple, because each display is somehow different. We need to create a new constructor for your.
Lets use the 192x96 constructor as a template.

Can you locate the correspondig file and line in you local copy of u8g2?

https://github.com/olikraus/u8g2/blob/f22a5a84209645f65c6ece0661c7f7518e7c8e9e/csrc/u8x8_d_st75256.c#L1689

Please remove the "8+" expression. Will it help?

vTaskCreate, 2022-12-27:
It's used the offset so string is moved through to bottom but characters still same, half cut.

olikraus, 2022-12-27:
Just to be sure: with flipmode example, the offset is still there and the frame at the LCD border is incorrect?

vTaskCreate, 2022-12-27:
The border(for Y) seems correct. The string located normally, but the character of string is cut half like it's not fitted as I showed with photo before.

olikraus, 2022-12-29:
I have created a new beta release 2.34.10
It includes the new constructor U8G2_ST75256_JLX16080, which is basically a copy of the 19296 constructor.

Please execute FlipMode.ino example (See example menu in Arduino IDE or [here](https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/FlipMode/FlipMode.ino)). The flip mode test example will cycle through eight modes. In all modes there must be a solid frame next to the border of the display.
Can you also send one or more pictures of the output?
 
You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

olikraus, 2023-02-22:
any feedback?

closed 2023-06-06 by olikraus