olikraus/u8g2 #2060: Wemos d1 mini (esp8266) 1.3 Oled + SD not working

X04N, 2022-12-23:
Hi, for some reason from the moment I initialize the SD card (SD.begin(4)) the oled display stops working.
I have an I2C 1.3" oled display connected to this pins: D1 - SCK and D2 - SDA.
l also have an SD module connected to the SPI pins ( SCK - D5, MISO - D6, MOSI - D7 and CS - D4).
I am using this as a constructor:
U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
I've been wrestling whith this for a couple of days and I can't seem to find a solution. My Google foo has also failed me.
The wiring has to be right, because independently both things work just fine, but when I try to initialize the SD card the display stops working and just stays blank.

olikraus, 2022-12-23:
Will another call to u8g2.begin() help?

X04N, 2022-12-23:
Thank you for the fast reply!
No, sadly it does not help.
I've tried moving the call to u8g2.begin() to the end of the setup() (after the SD.begin) nad it did not help.
I also tried adding it at the beginning of the loop and it didn't solve it either.
I am at a complete loss.
Again, thanks

olikraus, 2022-12-23:
In most of the similar cases, the other SPI lib reconfigures the SPI bus in such a way, that the display is not able to work anymore. You could try to restore the state of the SPI object to it's default state.

X04N, 2022-12-23:
Ok, I stumbled into a weird solution...
For some reason changing "SD.begin(4)" to "SD.begin(D4)" seems to have solved the problem.
Not sure why but well... thanks for the help!

closed 2023-02-22 by olikraus