olikraus/u8g2 #2062: passing u8g2 instance to a function

behnamsaberi, 2022-12-25:
hi and thanks for the great library . it almost covers all kind of graphical displays.
I have used it for cupule of months and it worked well. however in my recent project I need to pass the oled (ssd1309) instance to a function (the instructor is U8G2_SSD1309_128X64_NONAME2_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 52, /* data=*/ 51, /* cs=*/ 53, /* dc=*/ 48, /* reset=*/ 46); // arduino mega2560). in VC shows the u8g2 as "<error-type>" in pop-up message. I don't what type should be set as the other function's argument.
please guide me
thanks in advance

olikraus, 2022-12-25:
To Pass u8g2 as a reference argument, use U8G2&.
U8G2 is the parent class name.

behnamsaberi, 2022-12-26:
thanks for replaying , it works well.
and a suggestion to improve the library, as I've understood there is no function to print const string stored in PROGMEM and ram should be used . it would be great if there is an alternative
thanks again

olikraus, 2022-12-26:
U8g2 supports the standard print function. So all kind of string arguments supported by Arduino print is also supported by u8g2

closed 2023-02-22 by olikraus