olikraus/u8g2 #2063: Any guide to add new display LM6023, S1D15300 controller

olikraus, 2022-12-28:
### Discussed in https://github.com/olikraus/u8g2/discussions/2059

<div type='discussions-op-text'>

<sup>Originally posted by **allenchak** December 23, 2022</sup>
I have a GLCD display which is 128x64, similar to others. I would like to help to merge my LM6023 (using S1D15300 IC) repo to yours, may I know any guide / reference for adding a new.

Or just let me know if I provide anything could assist you to add it quickly, such as the initial commands, power control commands, etc.</div>

olikraus, 2022-12-28:
I started to add the S1D15300 controller to the u8g2 comparison list.
There is a significant difference in the contrast / e-volume setting.

allenchak, 2022-12-28:
Just let me know if you want to have a test or any help

olikraus, 2022-12-29:
I have created a new beta release 2.34.10
It includes the new constructor U8G2_S1D15300_LM6023, which is a copy of the U8G2_ST7565_JLX12864 constructor along with your updates.

Please execute FlipMode.ino example (See example menu in Arduino IDE or [here](https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/FlipMode/FlipMode.ino)). The flip mode test example will cycle through eight modes. In all modes there must be a solid frame next to the border of the display.

I do not expect much changes, but we can use this as a template for further improvements, so the question is, will this constructor work in the same why like your modified constructor? Can you sent any pictures?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

allenchak, 2022-12-30:
Thanks for your reply, and not expecting the beta release to be created shortly. 

I will test the beta release asap and get back the result. And thank you for your work. 

I will take a look if any issues exist in others examples.

allenchak, 2022-12-30:
seems it work on LOLIN D1mini -- https://youtu.be/o_P9gEUvCvM

I am testing other example, will update here.

Just one more minor changes in `u8x8_d_s1d15300.c`, update to `/* flipmode_x_offset = */ 4,`
 and here is my constructor

U8G2_S1D15300_LM6023_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 14, /* data=*/ 13, /* cs=*/ U8X8_PIN_NONE, /* dc=*/ 0, /* reset=*/ 4);

U8G2_S1D15300_LM6023_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ U8X8_PIN_NONE, /* dc=*/ 0, /* reset=*/ 4);

and the benchmark for flip mode using LOLIN D1mini @80MHz:

HW SPI: 11.332 seconds
SW SPI: 20.324 seconds

olikraus, 2022-12-30:
Thanks for the video. It looks like that the screen is mirrored in F1 mode. 
Can you test some changes for this?
https://github.com/olikraus/u8g2/blob/0e0ec4ee63e59eb58345f0d3d34f2e6435bc6348/csrc/u8x8_d_s1d15300.c#L68-L74
Either 0xa1 has to be 0xa0 or 0xc8 has to be 0xc0.
What will fix the mirrored display in F1 mode?

allenchak, 2022-12-31:
After some changes, the final version of the settings is as follows:

- /* flipmode_x_offset = */ 0
- `u8x8_d_s1d15300_flip1_seq` identical to `u8x8_d_s1d15300_flip0_seq`

video: https://youtu.be/v8mpbW3hi-U

olikraus, 2022-12-31:
The purpose of the flip mode is to have a 180 degree rotation, so ideally it should be different. Using the same sequence for flipmode will deactivate the same. I will check the datasheet later.

olikraus, 2022-12-31:
Datasheet: https://github.com/allenchak/LM6023/blob/master/Datasheet/S1D15300.pdf

After reading the datasheet, i think it should be like this:

/* flipmode_x_offset = */ 4

```
 static const uint8_t u8x8_d_s1d15300_flip1_seq[] = { 
   U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */ 
   U8X8_C(0x0a0),				/* segment remap a0/a1*/ 
   U8X8_C(0x0c8),				/* c0: scan dir normal, c8: reverse */ 
   U8X8_END_TRANSFER(),             	/* disable chip */ 
   U8X8_END()             			/* end of sequence */ 
 }; 
```

Can you sent a picture how this will look for F1 mode? 
BTW: You could also use u8x8 flip mode example, which is a little bit less confusing.

allenchak, 2022-12-31:
It seems something is wrong. 
If I set `/* flipmode_x_offset = */ 0`, the flip mode 1 will be aligned to the left, as same as flip mode 0
[image]
[image]

olikraus, 2022-12-31:
well, yes, but the text orientation should be rotated by 180 degree. I mean, the text should be next to the display pin array for flip mode 1.

How will this look like:

/* flipmode_x_offset = */ 4

```
 static const uint8_t u8x8_d_s1d15300_flip1_seq[] = { 
   U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */ 
   U8X8_C(0x0a1),				/* segment remap a0/a1*/ 
   U8X8_C(0x0c0),				/* c0: scan dir normal, c8: reverse */ 
   U8X8_END_TRANSFER(),             	/* disable chip */ 
   U8X8_END()             			/* end of sequence */ 
 }; 
```

Thanks for all the testing. There is always a kind of reverse engineering because datasheets are never that clear...

allenchak, 2023-01-01:
I have tested the 4 four combinations

- a1 & c0, 
- a0 & c0, 
- a1 & c8,
- a0 & c8

Unfortunately, all these settings are not making the expected result.

But I just found that you are referring to a **summary** datasheet. I am not sure is it any hints from the [complete datasheet](https://github.com/allenchak/LM6023/blob/master/Datasheet/s1d15605.pdf) on pages 123 to 163.

olikraus, 2023-01-01:
Ok, don't know how to continue from here. I think the display now works as expected in flip mode 0, but it will not show flip mode 1 correctly. 
This is not a big issue, we could just keep this as it is.

closed 2023-02-22 by olikraus